nickg/nvc #690: Records in 2D arrays causes sim to crash (SEGV_MAPERR)

nicholasjoukh, 2023-05-09:
[nvc_2d_array_test.zip](https://github.com/nickg/nvc/files/11425899/nvc_2d_array_test.zip)

I've written a small project that reproduces the failure. It's got a 2d array of a record that's got a single field. The test tries to output a filled instance of the 2D array in a module, and assert if the entries are correct in a test.

Result of running the project:
```
nvc -r tb_top

*** Caught signal 11 (SEGV_MAPERR) [address=0xffffffffffffff98, ip=0x556bf4ba5bd5] ***

[0x556bf4ae413d] ../src/util.c:850 signal_handler
[0x7f7b4aa9051f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x556bf4ba5bd5] ../src/rt/model.c:1537 split_nexus.lto_priv.0
[0x556bf4ba7281] ../src/rt/model.c:3578 x_map_signal
[0x556bf4bbf7d0] ../src/jit/jit-exits.c:699 __nvc_do_exit
[0x7f7b486354ea] (/home/nicholasj/work/nvc_test/work/_WORK.TB_TOP.elab.so) WORK.TB_TOP.DUT
[0x556bf4b9eefe] ../src/jit/jit-core.c:626 jit_fastcall
[0x556bf4ba39bb] ../src/rt/model.c:980 reset_scope.lto_priv.0
[0x556bf4ba359c] ../src/rt/model.c:991 reset_scope.lto_priv.0
[0x556bf4ba359c] ../src/rt/model.c:991 reset_scope.lto_priv.0
[0x556bf4aebe7e] ../src/rt/model.c:2126 model_reset
[0x556bf4aebe7e] ../src/rt/model.c:648 run
[0x556bf4ae893b] ../src/nvc.c:1494 process_command
[0x556bf4ae07b6] ../src/nvc.c:1618 main
```

nickg, 2023-05-09:
Thanks for the bug report but the zip file seems to be empty?

nicholasjoukh, 2023-05-09:
Oh yup it was,  sorry 🙏   I've reuploaded it in the original comment as well. 
[nvc_2d_array_test.zip](https://github.com/nickg/nvc/files/11436677/nvc_2d_array_test.zip)

nickg, 2023-05-10:
Should be fixed now, although your test case needs a `wait for 0 ns` or similar in `test_process` in order to pass.

closed 2023-05-10 by nickg