nickg/nvc #696: IEEE Synopsys STD_LOGIC_MISC is missing from 08 library

nicholasjoukh, 2023-05-19:
Can this be added please? xsim has STD_LOGIC_MISC in it's 08 library

nickg, 2023-05-19:
They must have modified the source because the original std_logic_misc won't compile in 2008 mode due to some of the changes made to the `std_logic_vector`/`std_ulogic_vector` subtyping arrangement. E.g. the following declares two separate overloads in -1993 but are duplicates in -2008:

```
    function Sense (V: STD_ULOGIC_VECTOR; vZ, vU, vDC: STD_ULOGIC) 
    					   return STD_LOGIC_VECTOR;
    function Sense (V: STD_ULOGIC_VECTOR; vZ, vU, vDC: STD_ULOGIC) 
    					   return STD_ULOGIC_VECTOR;
```

I guess that's why I skipped it originally. Does xsim provide the source code to their std_logic_misc?

nicholasjoukh, 2023-05-21:
> 

yeah it's in the Vivado install directory; `Xilinx/Vivado/2019.2/data/vhdl/2008/IEEE\ Synopsys/std_logic_misc.vhd`. Odd, there's no difference between their 08 and 93 std_logic_misc 😐

> They must have modified the source because the original std_logic_misc won't compile in 2008 mode due to some of the changes made to the `std_logic_vector`/`std_ulogic_vector` subtyping arrangement. [...]

nicholasjoukh, 2023-05-22:
We're thinking this could be Xilinx messing up function overloads. We've seen xsim allow bad overloads, so we'll just move to not using `std_logic_misc`. Thanks for the help!

closed 2023-05-22 by nicholasjoukh

nickg, 2023-05-22:
I looked at GHDL and it's compiling that package for 2008 using pre-processing directives to skip the duplicate definitions so I did the same thing in the commit above. Could you try again with the latest master branch?

nicholasjoukh, 2023-05-23:
> I looked at GHDL and it's compiling that package for 2008 using pre-processing directives to skip the duplicate definitions so I did the same thing in the commit above. Could you try again with the latest master branch?

oh nice it stopped complaining about it. Thanks!