nickg/nvc #700: String with leading NUL characters doesn't get printed in report

bpadalino, 2023-05-24:
```vhdl
use std.textio.all ;

entity test is
end entity ;

architecture arch of test is

begin

    tb : process
        variable s : string(1 to 4) ;
    begin
        s(3 to 4) := "Hi" ;
        report s ;
        write(output, s) ;
        std.env.stop ;
    end process ;

end architecture ;
```

I'd expect the output of the report to be the `NUL` characters followed by "Hi", but it isn't.  The call to `write` does the correct thing.

```
$ nvc --std=2008 -a test.vhdl -e test -r test &> file
$ hexdump -C file 
00000000  2a 2a 20 4e 6f 74 65 3a  20 30 6d 73 2b 30 3a 20  |** Note: 0ms+0: |
00000010  52 65 70 6f 72 74 20 4e  6f 74 65 3a 20 0a 09 20  |Report Note: .. |
00000020  2f 74 6d 70 2f 62 6c 61  68 2f 74 65 73 74 2e 76  |/tmp/blah/test.v|
00000030  68 64 6c 3a 31 34 0a 2a  2a 20 4e 6f 74 65 3a 20  |hdl:14.** Note: |
00000040  30 6d 73 2b 30 3a 20 53  54 4f 50 20 63 61 6c 6c  |0ms+0: STOP call|
00000050  65 64 0a 20 20 20 50 72  6f 63 65 64 75 72 65 20  |ed.   Procedure |
00000060  53 54 4f 50 20 5b 5d 0a  09 20 2e 2e 2f 6c 69 62  |STOP [].. ../lib|
00000070  2f 73 74 64 2e 30 38 2f  65 6e 76 2d 62 6f 64 79  |/std.08/env-body|
00000080  2e 76 68 64 3a 33 30 0a  20 20 20 50 72 6f 63 65  |.vhd:30.   Proce|
00000090  73 73 20 3a 74 65 73 74  3a 74 62 0a 09 20 2f 74  |ss :test:tb.. /t|
000000a0  6d 70 2f 62 6c 61 68 2f  74 65 73 74 2e 76 68 64  |mp/blah/test.vhd|
000000b0  6c 3a 31 36 0a 00 00 48  69                       |l:16...Hi|
000000b9
```

... `ghdl` prints what I expect:

```
$ ghdl -r --std=08 test | hexdump -C
00000000  74 65 73 74 2e 76 68 64  6c 3a 31 34 3a 39 3a 40  |test.vhdl:14:9:@|
00000010  30 6d 73 3a 28 72 65 70  6f 72 74 20 6e 6f 74 65  |0ms:(report note|
00000020  29 3a 20 00 00 48 69 0a  00 00 48 69 73 69 6d 75  |): ..Hi...Hisimu|
00000030  6c 61 74 69 6f 6e 20 73  74 6f 70 70 65 64 20 40  |lation stopped @|
00000040  30 6d 73 0a                                       |0ms.|
00000044
```

bpadalino, 2023-05-24:
That sounds reasonable to me.

nickg, 2023-05-26:
It prints this now:

```
** Note: 0ms+0: Report Note: ␀␀Hi
```

commit 264cbb8b (2023-05-26): Handle leading NUL characters in report message. Fixes #700
closed 2023-05-26 by nickg