nickg/nvc #703: GTKW output

amb5l, 2023-05-26:
NVC's GTKW output inserts a comment before the signals associated with a particular leaf in the hierarchy (as well as cycling the colour). The hierarchy levels are separated by a "." (period). Please could they be separated and also followed by a forward slash? (Or a user defined character?)

It also inserts a comment before members of a record, naming the record signal - but the record signal name is prepended by the hierarchy level, which is a little confusing. Could the hierarchy level be dropped? Separating multiple levels of record with a period makes sense. Could the record name(s) be followed by a ":" to visually distinguish them?

It would also be nice to have a visual break between the last signal of a record, and the next (non-record) signal. Could NVC insert a blank line (comment) in this case?

Something like:

```
root/ <-- trailing slash = start of hierachy level
my_signal_1
my_signal_2
my_rec_1:  <-- trailing colon = start of record 
member_1
member_2
 <-- empty (blank comment) = end of record, if non-record signals follow
my_signal_3
axi:
axi.aw:
addr
len
axi.w:
data
valid
ready
root/dut/
my_signal_4
```

nickg, 2023-05-29:
Could you try the latest master branch to see if that does what you want?

amb5l, 2023-06-02:
That's perfect. Thankyou very much.

closed 2023-06-02 by nickg