nickg/nvc #704: Signals missing from waveform

amb5l, 2023-05-26:
When I simulate my design, and write a FST waveform, some signals are missing when I look at the waveform in GTKWave3.3.111.

To reproduce,

- checkout [this commit](https://github.com/amb5l/tyto2/commit/5ce4a724e974e6b6a33277a74141d862733fd582)
- run `make nvc gtkwave` in `test/tb_axi4_a32d32_srw32`
- the source will be compiled and the simulation will run, exiting on an AXI4 protocol error
- run GTKWave: `gtkwave .nvc/sim.fst .nvc/sim.gtkw`
- look at GTKWave's hierarchy pane - under `dut` you will see records `axi4_so` and `sim_axi4`, but `axi4_si` is missing...

Note that I created the `sim_axi4` signal to debug this issue. It's derived from both `axi4_so` and `axi4_si`, so they are working, but `axi4_si` is not being dumped to the waveform.

nickg, 2023-05-28:
I tried to run this but got an error:

```
~/src/tyto2/test/tb_axi4_a32d32_srw32 $ make
makefile:29: /home/nick/src/tyto2/submodules/make-fpga/make-fpga.mak: No such file or directory
make: *** No rule to make target '/home/nick/src/tyto2/submodules/make-fpga/make-fpga.mak'.  Stop.
```

I think your problem is probably caused by port collapsing: for a simple input association like `axi4_si => axi4_si` the `axi4_si` in the upper scope will be merged with the `axi4_si` in the lower scope so after elaboration there is only one signal. Proprietary tools do this too but they may also turn it off for debugging and/or hide it from the user in the waveform display by making a copy of the signal data. Not sure if FST has a way to represent this, I'll have a look.

amb5l, 2023-05-29:
That error would happen if submodules are not initialized - let me know if that's not the case here.

Re: port collapsing - the sim_axi4 signal is a dead end signal (nothing listens to it) that exists to make waveform display consistent (different tools display record member names in different ways). Presumably signal collapsing is not a thing in NVC?

nickg, 2023-06-01:
I found FST has an "alias" feature which does almost exactly what I want here, so you can say a signal in one scope is identical to a signal in a different scope. However it only works for scalars and not for records so it doesn't help for your specific case (because records have a whole FST scope for themselves, they're not a single variable).

FWIW Modelsim/Questa does the same thing where it collapses ports by default and doesn't show them in the signals view unless you pass `+acc` when compiling. I could add a new elaboration option `--no-collapse` or similar if this would be useful for debugging.

> That error would happen if submodules are not initialized - let me know if that's not the case here.

Yes that was it, thanks. I reused an earlier checkout of that repository and didn't realise I needed to re-init the submodules.

> 
> Presumably signal collapsing is not a thing in NVC?

No, it won't get rid of signals if they're never used.

amb5l, 2023-06-02:
> I could add a new elaboration option `--no-collapse` or similar if this would be useful for debugging.

Yes please!

nickg, 2023-06-04:
I've implemented this now so if you do `nvc -e --no-collapse ...` the missing signals should appear in the waveform dump, although the simulation will run a tad slower. Could you give this a try?

amb5l, 2023-06-05:
That works. Thanks!

closed 2023-06-05 by nickg