nickg/nvc #709: Alias of an external name causes a segmentation fault

yikkrubz, 2023-06-09:
Analyzing the file B_pkg.vhd with a current version of nvc causes a segmentation fault. The example works as expected when the code is compiled by Questa.

nvc --std=08 -a A.vhd 
nvc --std=08 -a B_pkg.vhd 

*** Caught signal 11 (SIGSEGV) [address=0x0, ip=0x10788f396] ***

[0x1077c6c35] signal_handler+0x405
[0x7ff81ddaedfc] (/usr/lib/system/libsystem_platform.dylib) _sigtramp+0x1c
[0x10788f397] lower_expr+0x1697
[0x1078ab5ba] lower_decl+0x65a
[0x1078850d6] lower_decls+0x156
[0x107885d45] lower_decls+0xdc5
[0x1078c0d0d] lower_standalone_unit+0x23d
[0x107874811] analyse_vhdl+0x181
[0x1077becf1] process_command+0x2041
[0x1077bc53a] main+0x4aa

nvc --version
nvc 1.10-devel (1.9.0.r202.g88a72389) (Using LLVM 16.0.5)
macos version 12.6.6
[external_name.zip](https://github.com/nickg/nvc/files/11706370/external_name.zip)

nickg, 2023-06-11:
Fixing this is dependent on some large refactoring that I plan to do. It will be supported eventually but for now I just added a more helpful error message.

nickg, 2023-10-29:
This should be fixed now, sorry for the long delay. But I think your example code is not valid as a relative pathname should always be enclosed within a concurrent region, see section 8.7 of the 2008 LRM:

> For a relative pathname, the innermost concurrent region is initially identified, where a concurrent region is defined recursively to be
—A block declarative region (including an external block and any block equivalent to a
generate statement), or
—A package declarative region (including a generic-mapped package equivalent to a package instantiation) declared immediately within a concurrent region.

My understanding is that only a package that is declared or instantiated within a block declarative region (i.e. within an architecture) counts as a "concurrent region" here. And then the next paragraph says:

> It is an error when evaluating the external name if, at any stage, there is no such containing declarative region, or if the containing declarative region is the declarative region of an uninstantiated package.

When changed to an absolute pathname it will work with the latest master branch:

```vhdl
    alias d is << signal .C.DUT.d : std_logic_vector >> ;
```

commit 38ea3aea (2023-10-29): Elaborate external names as callback from code generation

Fixes #709
closed 2023-10-29 by nickg