nickg/nvc #715: Failing due to out of date architecture that is not the MRA architecure

JimLewis, 2023-06-17:
If I go through the sequence
```
# Build and run test case Uart1_Rx
analyze UartTbPkg.vhd
analyze UartTx.vhd
analyze UartRx.vhd
analyze TestCtrl_e.vhd
analyze ../../Tb2_other_src/TbMemIO_Struct.vhd
analyze TestCtrl_Uart1_Rx.vhd
simulate TbMemIO
# Build and run test case Uart2_Rx_ErrorInjection
analyze UartTbPkg.vhd
analyze UartTx.vhd
analyze ../Tb2_Entity_1_UartTx/UartRx.vhd
analyze ../Tb2_Entity_1_UartTx/TestCtrl_e.vhd
analyze ../../Tb2_other_src/TbMemIO_Struct.vhd
analyze TestCtrl_Uart2_Rx_ErrorInjection.vhd
simulate TbMemIO
```

The second simulate results in the following elaboration error.   Note it is complaining that Uart1_Rx architecture is out of date.   Indeed it is out of date.  However it is not the most recently analyzed architecture, so it is not the one that should be loaded for simulation and hence, the design should elaborate and select architecture Uart2_Rx_ErrorInjection
```
nvc --std=08 -H 128m --work=Tb2_Entity:C:/tools/sim_temp/nvc-1.10-devel/TB2_ENTITY.08 -L C:/tools/sim_temp/nvc-1.10-devel -e TbMemIO
Elaborate Error: ** Fatal: C:\tools\sim_temp\nvc-1.10-devel\TB2_ENTITY.08\TB2_ENTITY.TESTCTRL-UART1_RX: design unit depends on TB2_ENTITY.TESTCTRL with checksum abfd7d33 but the current version in the library has checksum b06d321e
Elaborate Error: ** Note: this usually means TB2_ENTITY.TESTCTRL-UART1_RX is outdated and needs to be reanalysed
Simulation Finish time 18:28:09, Elasped time: 0:00:00 
```

I think I am seeing this issue go away when I use a configuration declarations to run in the scenario, but I am not 100% certain the scenario is the same.

For what it is worth, these test cases pass on GHDL, Questa, RivieraPRO, and ActiveHDL.   That said, much to my amusement I have seen a 5th tool actually find a coding bug that others have missed.

JimLewis, 2023-06-17:
Similar to #710, except this is fatal.   I would not have even noticed a warning.  :)

nickg, 2023-06-17:
This should be fixed now, could you test again? There are other issues with stale design units in the library which are covered by #710.

closed 2023-08-13 by nickg