nickg/nvc #725: Fatal: uarray must have at least one dimension

avelure, 2023-06-20:
The following gives a fatal "uarray must have at least one dimension"
It passes in modelsim and GHDL.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test is
  type t_slv_array is array (natural range <>) of std_logic_vector;
  subtype t_word              is std_logic_vector(15 downto 0);
  subtype t_word_array        is t_slv_array(open)(t_word'range);
  subtype t_addr              is natural range 1 to 30;
  type t_data is array (t_addr) of t_word_array(0 to 31);
end package test;
```

```
Name       WORK.TEST."="(16WORK.TEST.T_DATA16WORK.TEST.T_DATA)B
Kind       function
Context    WORK.TEST
Blocks     8
Registers  46
Types      18
Variables  2
  i40                                   // #, temp
  i42                                   // #, temp
Result     0..1
Parameters 3
  r0    context                         // P<WORK.TEST>
  r1    L                               // [*] : 0..8 => 0..8
  r2    R                               // [*] : 0..8 => 0..8
Begin
   0: r3 := unwrap r1                   // @<0..8> => 0..8
      r4 := unwrap r2                   // @<0..8> => 0..8
      r5 := uarray len r1 dim 0         // #
      r6 := uarray len r2 dim 0         // #
      r7 := const 0                     // # => 0
      i40 := store r7
      r8 := const 1                     // # => 1
      r9 := const 32                    // # => 32
      r10 := const 0                    // -2^31..2^31-1 => 0
      r11 := const 31                   // -2^31..2^31-1 => 31
      r12 := const 0                    // 0..1 => 0
      r13 := const 15                   // -2^31..2^31-1 => 15
      r14 := const 1                    // 0..1 => 1
      r15 := null                       // @<#>
      r16 := wrap r15 [r10 r11 r12, r13 r10 r14] // [*, *] : # => @<#>
      r17 := const 16                   // # => 16
      r18 := const 512                  // # => 512
      r19 := cmp r5 == r6
      cond r19 then 2 else 1
   1: Empty basic block
   2: r20 := load i40                   // #
      r21 := cmp r20 == r5
      cond r21 then 4 else 3
   3: r22 := mul r20 * r18              // # => -2^63..2^63-1
      r23 := const 1                    // # => 1
      r24 := add r20 + r23              // # => -9223372036854775807..2^63-1
      i40 := store r24
      r25 := cmp r24 == r5
      r26 := array ref r3 offset r22    // @<0..8> => 0..8
      r27 := array ref r4 offset r22    // @<0..8> => 0..8
      r28 := const 32                   // # => 32
      r29 := const 0                    // # => 0
      i42 := store r29
      r30 := const 0                    // -2^31..2^31-1 => 0
      r31 := const 31                   // -2^31..2^31-1 => 31
      r32 := const 0                    // 0..1 => 0
      r33 := const 15                   // -2^31..2^31-1 => 15
      r34 := const 1                    // 0..1 => 1
      r35 := null                       // @<#>
      r36 := wrap r35 [r30 r31 r32, r33 r30 r34] // [*, *] : # => @<#>
      r37 := const 16                   // # => 16
      jump 5
   4: Empty basic block
   5: r38 := load i42                   // #
      r39 := cmp r38 == r28
      cond r39 then 7 else 6
   6: r40 := mul r38 * r37              // # => -2^63..2^63-1
      r41 := const 1                    // # => 1
      r42 := add r38 + r41              // # => -9223372036854775807..2^63-1
      i42 := store r42
      r43 := cmp r42 == r28
      r44 := array ref r26 offset r40   // @<0..8> => 0..8
      r45 := array ref r27 offset r40   // @<0..8> => 0..8
      invalid := wrap r44 []     <----
   7: Empty basic block

** Fatal: uarray must have at least one dimension
[00007FF7098E2FBF]
[00007FF709988855]
[00007FF7098E21D2]
[00007FF70996F892]
[00007FF7099786FD]
[00007FF7099531DB]
[00007FF709966041]
[00007FF709965EC6]
[00007FF709966386]
[00007FF70996CD38]
[00007FF7098EC20D]
[00007FF7099ED4C0] vhpi_control+0x24800
[00007FF7098E12EE]
[00007FF7098E1406]
[00007FFFFF437614] BaseThreadInitThunk+0x14
[00007FFFFF9426F1] RtlUserThreadStart+0x21
```

avelure, 2023-06-20:
I noticed now this is fixed on latest master, so it can be closed unless you collect regression test cases.

nickg, 2023-06-20:
I added it as a test case, thanks.

closed 2023-06-20 by nickg