nickg/nvc #727: Signature matching fails when provided parameter is the sliced return value from a function without parameters

avelure, 2023-06-21:
I managed to recreate the main issue from #726
It looks like the code that tries to match the signature thinks that I'm providing a parameter to get_val, but in reality I'm slicing the return value. If I'm using the temporary variable v_test it works.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package types_pkg is
  type t_slv_array is array (natural range <>) of std_logic_vector;
  subtype t_word              is std_logic_vector(15 downto 0);
  subtype t_word_array        is t_slv_array(open)(t_word'range);
  subtype t_addr              is natural range 1 to 30;
  type t_data is array (t_addr) of t_word_array(0 to 31);
end package types_pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.types_pkg.all;

package protected_pkg is
  type t_config_interface is protected
    impure function get_val return t_data;
  end protected t_config_interface;
end package protected_pkg;
package body protected_pkg is
  type t_config_interface is protected body
    variable v_data : t_data := (others => (others => (others => '0')));
    impure function get_val return t_data is
    begin
      return v_data;
    end function get_val;
  end protected body t_config_interface;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.protected_pkg.all;

package shared_variables_pkg is
  shared variable shared_config : t_config_interface;
end package shared_variables_pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.types_pkg.all;
use work.shared_variables_pkg.all;

package test_pkg is
  impure function randomize(constant val : t_word_array) return t_word_array;
  impure function randomize(constant sub_address : unsigned) return t_word_array;
end package test_pkg;

package body test_pkg is
  impure function randomize(constant val : t_word_array) return t_word_array is
  begin
    return val;
  end function;
  impure function randomize(constant sub_address : unsigned)
    return t_word_array is
    variable v_test : t_word_array(0 to 0);
  begin
    v_test := shared_config.get_val(to_integer(sub_address));
    v_test := randomize(v_test);
    return randomize(shared_config.get_val(to_integer(sub_address)));
  end function randomize;
end package body;
```
```
$ nvc --std=08 -a test.vhd
** Error: no matching subprogram RANDOMIZE [T_DATA return T_WORD_ARRAY]
    > C:\proj\public\minimal\test_uarray.vhd:67
    |
 67 |     return randomize(shared_config.get_val(to_integer(sub_address)));
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

nickg, 2023-06-21:
This should be fixed by the commit above. Could you test again?

avelure, 2023-06-22:
Thanks for quick fix, it works now :)