nickg/nvc #733: Type visibility for homographs is too lax

bpadalino, 2023-07-06:
I recently came across a difference between `nvc` and `ghdl`, but the behavior of `nvc` matched Riviera-PRO so I thought `nvc` was correct.  The [issue](https://github.com/ghdl/ghdl/issues/2463) revealed that the LRM specifically states that the `ghdl` behavior is correct.

```vhdl
package outer is

    type a is record
        x : bit_vector ;
    end record ;

    package inner is
      generic (
        WIDTH : positive := 1
      ) ;
        subtype a is a(x(WIDTH-1 downto 0)) ;
    end package ;

end package ;
```

... according to the LRM, right after `subtype a is` in the inner package, the visibility of `a` is hidden so that subtype declaration is illegal.  The subtype declaration should either use an alias such that the homograph `a` for the outer package is still visible using a different name, or use the expanded name `outer.a` as @tgingold pointed out.

Also, while writing this up, removing the `generic` from `inner` and modified as such:

```vhdl
package outer is

    type a is record
        x : bit_vector ;
    end record ;

    package inner is
        subtype a is a(x(7 downto 0)) ;
    end package ;

end package ;
```

... causes a different error:

```
$ nvc --std=2008 -a visibility.vhdl
** Fatal: tree kind T_PACKAGE does not have item I_REF
   > visibility.vhdl:7
   |
 7 |     package inner is
   |     ^
[0x55c7888c7e68] ../src/object.c:213 object_lookup_failed
[0x55c7888c97df] ../src/tree.c:970 lower_decls.lto_priv.0.cold
[0x55c78893d384] ../src/lower.c:11135 analyse_file
[0x55c7888d1e9e] ../src/nvc.c:229 analyse
[0x55c7888d1e9e] ../src/nvc.c:1557 process_command
[0x55c7888cd3fa] ../src/nvc.c:1685 main
```

I am unsure if the current behavior might be desirable with the `--relaxed` since at least Riviera-PRO allows it.

nickg, 2023-07-07:
I noticed GHDL doesn't error on this:

```vhdl
entity visibility8 is
end entity;

architecture test of visibility8 is
    constant c1 : integer := 1;
begin

    p1: process is
        constant c1 : integer := c1;    -- Error
    begin
        wait;
    end process;

end architecture;
```

Shouldn't this rule also apply to the declaration of `c1` inside the process?

nickg, 2023-07-07:
I've implemented this now but I'm unsure why GHDL doesn't error for the other cases in [test/parse/visibility8.vhd](https://github.com/nickg/nvc/blob/master/test/parse/visibility8.vhd).

bpadalino, 2023-07-07:
Seems like some inconsistent behavior.  I submitted [issue #2465](https://github.com/ghdl/ghdl/issues/2465) with `ghdl`.

bpadalino, 2023-07-07:
Unsure if you care about this at all or not, but GHDL does seem to allow **most** of them through when using `-frelaxed`.  It isn't necessary to do that immediately, and I am not sure how difficult it is to switch it on/off if things are relaxed.

As I mentioned with the `ghdl` issue, Riviera-PRO allows it by default.

nickg, 2023-07-07:
I think I'll leave it as an error for now rather than introduce the complexity of having different behaviour in relaxed mode.

closed 2023-07-07 by nickg