nickg/nvc #736: Tests failing when building NVC with JIT

Blebowski, 2023-07-07:
When I configure NVC and build it with JIT, many tests fail:

```
../configure --enable-jit
make 
sudo make install
make check
```

I get:
```
68 failures!
FAIL: bin/run_regr
====================================================
1 of 2 tests failed
Please report to https://github.com/nickg/nvc/issues
====================================================
make[2]: *** [Makefile:3458: check-TESTS] Error 1
make[1]: *** [Makefile:3723: check-am] Error 2
make: *** [Makefile:3725: check] Error 2
```

Most of the stuff seems to be related to "bounds":
```
      bounds39 : failed (no match line 1)
1ns+0: value length 9 does not match subtype BV10 length 10
/WORK/nvc/build/bin/nvc --std=1993 -a /WORK/nvc/test/regress/bounds39.vhd -e bounds39 -O0
/WORK/nvc/build/bin/nvc --std=1993 -r bounds39
** Fatal: (init): cannot generate JIT IR for WORK.BOUNDS39
[0x5613d90eb653] ../src/diag.c:971 diag_femit
[0x5613d9040265] ../src/util.c:583 fatal_trace
[0x5613d91005dc] ../src/jit/jit-core.c:394 jit_fill_irbuf
[0x5613d9118c31] ../src/jit/jit-interp.c:1076 jit_interp
[0x5613d90faf0e] ../src/jit/jit-core.c:610 jit_fastcall
[0x5613d91003dc] ../src/rt/model.c:1066 reset_scope.lto_priv.0
[0x5613d90fffbc] ../src/rt/model.c:1077 reset_scope.lto_priv.0
[0x5613d903d482] ../src/rt/model.c:2208 model_reset
[0x5613d903d482] ../src/rt/model.c:712 run
[0x5613d903a27b] ../src/nvc.c:1561 process_command
[0x5613d9035eda] ../src/nvc.c:1685 main
```
Is this expected?

nickg, 2023-07-07:
I think that mode is probably broken at the moment due to some refactoring work. I'll fix it eventually (and also make it the default). You can also use the JIT mode by passing `--no-save -jit` when elaborating but it only works when doing elaboration and simulation in the same command (i.e. `nvc -e --no-save --jit tb -r`).

Blebowski, 2023-07-07:
Ok, thanks for explanation. I can keep the issue open as a tracker?

nickg, 2023-07-07:
Yes let's leave it open.

commit 1aea6115 (2023-07-10): Hack to allow elaborated unit in unit_registry_get. Fixes #736
closed 2023-07-10 by nickg