nickg/nvc #737: String literal is confused that it might be bit_vector

bpadalino, 2023-07-07:
Trying to write a `pprint` procedure using a `value_mirror` for 2019, and I came across this little bug.

```vhdl
package pprint is

    procedure pprint(variable x : in std.reflection.value_mirror) ;

end package ;

package body pprint is

    use std.textio.output ;
    use std.textio.line ;
    use std.textio.write ;
    use std.textio.writeline ;

    procedure pprint(variable x : in std.reflection.value_mirror) is
        variable l : line ;
    begin
        write(l, "pprint(") ;
        writeline(output, l) ;
        write(l, ")") ;
        writeline(output, l) ;
    end procedure ;

end package body ;
```

```
$ nvc --std=2019 -a pprint.vhdl 
** Error: type of string literal cannot be determined from the surrounding context
    > pprint.vhdl:17
    |
 17 |         write(l, "pprint(") ;
    |                  ^^^^^^^^^ could be BIT_VECTOR or STRING
    |
    = Note: context contains overload WRITE [LINE, BIT_VECTOR, SIDE, WIDTH]
         ../lib/std.19/textio.vhdl:125
    = Note: context contains overload WRITE [LINE, STRING, SIDE, WIDTH]
         ../lib/std.19/textio.vhdl:131
** Error: type of string literal cannot be determined from the surrounding context
    > pprint.vhdl:19
    |
 19 |         write(l, ")") ;
    |                  ^^^ could be BIT_VECTOR or STRING
    |
    = Note: context contains overload WRITE [LINE, BIT_VECTOR, SIDE, WIDTH]
         ../lib/std.19/textio.vhdl:125
    = Note: context contains overload WRITE [LINE, STRING, SIDE, WIDTH]
         ../lib/std.19/textio.vhdl:131
```

Obviously `"pprint("` and `")"` are not `bit_vector` types.

Note that removing the 2019 stuff doesn't make a difference, all the other standards have the same confusion.

diogratia, 2023-07-08:
I don't have a copy of the-2019 standard on my iPad but -2008 9.3.2 Literals tells us:

>String and bit string literals are representations of one-dimensional arrays of characters. The type of a string or bit string literal shall be determinable solely from the context in which the literal appears, excluding the literal itself but using the fact that the type of the literal shall be a one-dimensional array of a character type. The lexical structure of string and bit string literals is defined in Clause 15.

See also 12.5 The context of overload resolution.

You'd find if you don't examine the value of the literal which overload of the write procedure shown is intended is ambiguous.

bpadalino, 2023-07-08:
@diogratia You're right - it seems that Riviera-PRO does some extra looking into the string itself which isn't required by the LRM.  Questa and GHDL seem to both get confused between `string` and `bit_vector` just as `nvc` does.

Closing the issue.

closed 2023-07-08 by bpadalino

nickg, 2023-07-08:
I think that's why the `swrite` alias was added in 2008 LRM along with a note:

```vhdl
alias SWRITE is WRITE [LINE, STRING, SIDE, WIDTH];
```

> NOTE 3—A call to a WRITE procedure with a string literal for the VALUE parameter is ambiguous, as the string could be interpreted as a value of type STRING or type BIT_VECTOR. If the intention is to write a value of type STRING, the alias SWRITE can be called without ambiguity.