nickg/nvc #746: 1.10.1: Fatal: directory lib/ieee.08 already exists and is not an NVC library

yurivict, 2023-07-29:
Build fails:
```
--- lib/ieee.08/IEEE.MATH_REAL ---
** Fatal: directory lib/ieee.08 already exists and is not an NVC library
*** [lib/ieee.08/IEEE.MATH_REAL] Error code 1
```

clang-15
FreeBSD 13.2

nickg, 2023-07-29:
I can't reproduce this. I tried manually updating the version in `/usr/ports/cad/nvc/Makefile` to 1.10.1 but it built ok. Maybe try disabling `-j` or use `MAKE=gmake`?

yurivict, 2023-07-29:
```-j 1``` helped, thanks.

But it looks like there is some dependency problem though.
It might depend on versions of some other packages. Maybe the gmake version.

m42uko, 2023-08-25:
I can confirm this also happens **sometimes** on Linux (ArchLinux, Linux kronii 6.4.11-arch2-1 #1 SMP PREEMPT_DYNAMIC Sat, 19 Aug 2023 15:38:34 +0000 x86_64 GNU/Linux) when using a high core count to compile. Seems to be some kind of race condition.

nickg, 2023-08-25:
I think the problem is that creating the library involves two steps: 1) making the directory and 2) creating the `_NVC_LIB` file within it which is used as a lock file.  If two processes attempt to concurrently create the same library it's possible for one of them to observe the intermediate state between 1 and 2 where the library exists but the lock file doesn't and then report this error. Should be fixed by the commit above.

commit 282eac59 (2023-08-25): Avoid race when checking if an existing directory is a library

Fixes #746
commit 143498e1 (2023-09-17): Avoid race when checking if an existing directory is a library

Fixes #746
closed 2023-08-25 by nickg