nickg/nvc #747: Fatal Error - Index out of bound check not performed

Anselmo95, 2023-07-31:
Hello,

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity e is
generic (
    L : natural := 1
);
end entity;

architecture test of e is
    signal s : std_logic_vector(L-1 downto 0);
begin

    process
    begin
        wait for 1 ns;
        s(10) <= '0'; -- no check done for out of bound
        s(12 downto 11) <= "11"; -- check is done here
        wait for 1 ns;
        report "SIMULATION ENDED" severity failure;
        wait;
    end process;

end architecture;
```

```
$ nvc -e e -g L=10; nvc -r e

*** Caught signal 11 (SEGV_MAPERR) [address=0x18, ip=0x560a0a901a38] ***

[0x560a0a839e29] ./../src/util.c:860 signal_handler
[0x7f48aaa4251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x560a0a901a38] ./../src/rt/model.c:1639 split_nexus
[0x560a0a9076f6] ./../src/rt/model.c:3553 x_drive_signal
[0x560a0a92a09f] ./../src/jit/jit-exits.c:697 __nvc_do_exit
[0x560a0a926292] ./../src/jit/jit-interp.c:854 interp_loop
[0x560a0a926292] ./../src/jit/jit-interp.c:1110 jit_interp
[0x560a0a912117] ./../src/jit/jit-core.c:614 jit_fastcall
[0x560a0a902bf6] ./../src/rt/model.c:952 reset_scope
[0x560a0a902b03] ./../src/rt/model.c:1086 reset_scope
[0x560a0a905935] ./../src/rt/model.c:2219 model_reset
[0x560a0a81c013] ./../src/nvc.c:732 run
[0x560a0a81e60d] ./../src/nvc.c:1696 process_command
[0x560a0a81ea0a] ./../src/nvc.c:1823 main
```

I did some digging into the code and I think the issue is inside the [emit_bounds_check](https://github.com/nickg/nvc/blob/ea457d5de2771095f4fcf64419b13073d006c29e/src/vcode.c#L5569) function.
For the first signal assigment, during elaboration, I can see that it enters the function above but it returns on the`static_ok` check.
I verified that the check is actually emitted for the second signal assigment, but this is as far as I could get.

nickg, 2023-07-31:
It's because it thought it already checked that index in `src/bounds.c`. Should be fixed now.

closed 2023-07-31 by nickg

Anselmo95, 2023-07-31:
Thanks for the qick fix !

commit f6401837 (2023-07-31): Array bounds check incorrectly optimised out. Fixes #747
commit 8f73ffa5 (2023-08-11): Array bounds check incorrectly optimised out. Fixes #747