nickg/nvc #750: fatal: invalid type kind T_ARRAY in range_of

mariuselv, 2023-08-09:
Fatal error when compiling Utility library in UVVM_3_BETA (https://github.com/UVVM/UVVM_3_BETA)

```
fatal: invalid type kind T_ARRAY in range_of
[0x102ea1aff] diag_femit+0x9c7
[0x102e10757] fatal_trace+0x5b
[0x102e5559f] range_of+0xe3
[0x102e7546b] lower_array_stride+0x117
[0x102e6e883] lower_array_total_len+0xbf
[0x102e74233] lower_default_value+0x1bf
[0x102e623cf] lower_decls+0x13e7
[0x102e64c57] lower_decls+0x3c6f
[0x102e6140f] lower_decls+0x427
[0x102e7f6ab] lower_standalone_unit+0x193
[0x102e57fa7] analyse_file+0xef
[0x102e0d35b] process_command+0x1587
[0x102e0b9cb] main+0x3cb
Error: Program ended with exit code 1
Failed to compile /UVVM_3_BETA/uvvm_util/src/global_signals_and_shared_variables_pkg.vhd!
```

mariuselv, 2023-08-09:
Fixed, coding error.

closed 2023-08-09 by mariuselv

nickg, 2023-08-09:
Could you share what it was? It shouldn't crash like that even on invalid code. I tried UVVM_3_BETA but ran into a different error which I'm fixing now.

mariuselv, 2023-08-09:
Sure.
The data_queue_pkg.vhd reported:
`UVVM_3_BETA/uvvm_util/src/data_queue_pkg.vhd:33:38: error: type of aggregate cannot be determined from the surrounding context`

Had to add a constant for the default value. Changed from 

 ```
 -- Declaration of storage
  subtype t_data_buffer is std_logic_vector(C_TOTAL_NUMBER_OF_BITS_IN_DATA_BUFFER - 1 downto 0);

  package protected_data_buffer_pkg is new work.protected_generic_types_pkg
    generic map(t_generic_element => t_data_buffer,
                c_generic_default => (others => '0'));
  use protected_data_buffer_pkg.all;

  shared variable shared_data_buffer : protected_data_buffer_pkg.t_prot_generic_array;
```

To 

```
  -- Declaration of storage
  subtype t_data_buffer is std_logic_vector(C_TOTAL_NUMBER_OF_BITS_IN_DATA_BUFFER - 1 downto 0);
  constant C_DATA_BUFFER_DEFAULT : t_data_buffer := (others => '0');

  package protected_data_buffer_pkg is new work.protected_generic_types_pkg
    generic map(t_generic_element => t_data_buffer,
                c_generic_default => C_DATA_BUFFER_DEFAULT);
  use protected_data_buffer_pkg.all;

  shared variable shared_data_buffer : protected_data_buffer_pkg.t_prot_generic_array;
```

In global_signals_and_shared_variables_pkg this fata error was reported

```
fatal: invalid type kind T_ARRAY in range_of
[0x10213daff] diag_femit+0x9c7
[0x1020ac757] fatal_trace+0x5b
[0x1020f159f] range_of+0xe3
[0x10211146b] lower_array_stride+0x117
[0x10210a883] lower_array_total_len+0xbf
[0x102110233] lower_default_value+0x1bf
[0x1020fe3cf] lower_decls+0x13e7
[0x102100c57] lower_decls+0x3c6f
[0x1020fd40f] lower_decls+0x427
[0x10211b6ab] lower_standalone_unit+0x193
[0x1020f3fa7] analyse_file+0xef
[0x1020a935b] process_command+0x1587
[0x1020a79cb] main+0x3cb
Error: Program ended with exit code 1
Failed to compile /UVVM_3_BETA/uvvm_util/src/global_signals_and_shared_variables_pkg.vhd!
```
 because of this package declaration - which was not in use and could be removed

```
  -- Protected type: string
  constant C_PROTECTED_STRING_DEFAULT : string := "";
  package protected_string_pkg is new work.protected_generic_types_pkg
    generic map(t_generic_element => string,
                c_generic_default => C_PROTECTED_STRING_DEFAULT);
  use protected_string_pkg.all;
```

Br.
Marius

nickg, 2023-08-09:
I've fixed the original "type of aggregate cannot be determined from the surrounding context" problem but this is another error in one of the other files that I am still working on.

nickg, 2023-08-11:
Should be fixed in the master branch now. By the way I think `uvvm_vvc_framework/script/compile_order.txt` is missing `../src/ti_vvc_framework_context.vhd`.