nickg/nvc #751: In nvc r1.9.0 or later, the value of the record_type field generated by function is not the expected value.

ikwzm, 2023-08-23:


nickg, 2023-08-24:
This is fixed on the master branch now. Thanks for the reproduction instructions.

closed 2023-08-24 by nickg

ikwzm, 2023-08-24:
Thank you!

We succeeded as follows.

```console
shell$ nvc --version
nvc 1.11-devel (1.10.0.r110.gc3127d46) (Using LLVM 10.0.0)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.

shell$ make test_8x0x0x0x0
sh analyze_libs.sh
nvc -L ./ -M 32M --work=WORK -e  image_stream_player_test_8x0x0x0x0
nvc -L ./ -M 32M --work=WORK -r  image_stream_player_test_8x0x0x0x0
 PARAM
   ELEM_BITS   = 8
   ATRB_BITS   = 3
   INFO_BITS   = 0
   SHAPE.C SIZE=1 LO=0 HI=0 DE=T DA=F
        .D SIZE=1 LO=0 HI=0 DE=F DA=F
        .X SIZE=1 LO=0 HI=0 DE=T DA=F
        .Y SIZE=1 LO=0 HI=0 DE=T DA=F
   DATA            SIZE=8 LO=0 HI=7
      ELEM_FIELD   SIZE=8 LO=0 HI=7
      ATRB_FIELD   SIZE=0 LO=8 HI=8
      ATRB_FIELD.C SIZE=0 LO=8 HI=8
      ATRB_FIELD.D SIZE=0 LO=8 HI=8
      ATRB_FIELD.X SIZE=0 LO=8 HI=8
      ATRB_FIELD.Y SIZE=0 LO=8 HI=8
      INFO_FIELD   SIZE=0 LO=9 HI=9
        25 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST ELEM_BITS=8  C=0   D=0   X=0   Y=0  
        35 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST.1.1   C_SIZE=32  D_SIZE=1   X_SIZE=1   Y_SIZE=1  
       375 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST.1.2   C_SIZE=1   D_SIZE=4   X_SIZE=3   Y_SIZE=3  
       755 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST.1.3   C_SIZE=3   D_SIZE=1   X_SIZE=20  Y_SIZE=20 
     12775 ns| MARCHAL < IMAGE_STREAM_PLAYER_TEST.1.4   C_SIZE=4   D_SIZE=4   X_SIZE=20  Y_SIZE=20 
  ***  
  ***  ERROR REPORT test_8x0x0x0x0
  ***  
  ***  [ MASTER ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
  ***  [ SLAVE ]
  ***    Error    : 0
  ***    Mismatch : 0
  ***    Warning  : 0
  ***  
** Note: 76796ns+0: Assertion Note: Simulation complete(success).
	 /mnt/d/ichiro/work/PipeWorkTest/src/test/vhdl/image_stream_models/image_stream_player_test_bench.vhd:290
```