nickg/nvc #754: Selective import of type from package causes operator matching error

avelure, 2023-09-01:
The code below fails to elaborate.
If I change the import to `use work.test_pkg.all;` it works ok.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type t_channel is (NOM, RED);
end package test_pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.test_pkg.t_channel;

entity test is
  port (
    current_active_channel : in  t_channel
  );
end entity test;

architecture beh of test is
begin
  
  process
  begin
    if current_active_channel = NOM then
      null;
    end if;
    wait;
  end process;

end architecture beh;
```
```
nvc --std=08 -a test_enum_compare.vhd
** Error: no matching operator "=" [T_CHANNEL, T_CHANNEL return BOOLEAN | BIT | STD_ULOGIC]
    > C:\proj\public\minimal\test_enum_compare.vhd:26
    |
 26 |     if current_active_channel = NOM then
    |        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

nickg, 2023-09-02:
It's another missed 2008 change. Fixed in ca2a94f7e.

closed 2023-09-02 by nickg

diogratia, 2023-09-04:
The 2008 Change to which Nick alludes:

[image]

The second paragraph of 5.1 in IEEE Std 1076-2008 changes the definition of operations of a type excluding predefined operators and removing the requirement that the operators are implicitly declared.

The change allows an implementation in an object oriented programming language to use class inheritance (see 5.2 Scalar types 5.2.1 General, 1st paragraph here) rather than implicit declaration. This in no way affects the ability to use pure functions implementing operator overload or for operations of a type while reducing implicit declaration overhead.

You could consider the *implicitly declared* in the quoted paragraph of 16.3 to be eligible for review and possible revision.