nickg/nvc #761: Overload Resolution Fails

JimLewis, 2023-09-18:
On DEV branch of OSVVM:
build $OsvvmLibraries
build $OsvvmLibraries/Common/TbParams/build.pro

NVC Error:
analyze TbModelParameters1.vhd
nvc --std=08 -H 128m --work=tb_ModelParams_Singleton:C:/tools/VHDL_LIBS/nvc-1.10-devel/TB_MODELPARAMS_SINGLETON.08 -L C:/tools/VHDL_LIBS/nvc-1.10-devel -a C:/SynthWorks/Dev/_osvvm/OsvvmLibraries/Common/TbParams/TbModelParameters1.vhd
** Error: ambiguous call to function GET
Error:      > C:\SynthWorks\Dev\_osvvm\OsvvmLibraries\Common\TbParams\TbModelParameters1.vhd:144
Error:      |
Error:  144 |       AffirmIfEqual(TbID, Get(Params, i), -i, "Get(Params, i) = " & to_string(-i)) ;
Error:      |                           ^^^^^^^^^^^^^^ use of name GET here
Error:      |
Error:      = Note: candidate GET [MODELPARAMETERSIDTYPE, NATURAL return INTEGER] from OSVVM_COMMON.MODELPARAMETERSSINGLETONPKG
Error:      = Note: candidate GET [MODELPARAMETERSIDTYPE, NATURAL return STD_LOGIC_VECTOR] from OSVVM_COMMON.MODELPARAMETERSSINGLETONPKG
AnalyzeError: See messages above in "analyze TbModelParameters1.vhd"

Should not be an error since the third parameter of AffirmIfEqual is an integer then the second parameter of AffirmIfEqual must also be an integer, and hence, the call to Get(Params, i) must be the one that returns an integer.

nickg, 2023-09-24:
Fixed on the master branch. I checked all the OSVVM tests now pass on the dev branch.

closed 2023-09-24 by nickg

JimLewis, 2023-09-24:
Thanks Nick.