nickg/nvc #769: ... but the current version in the library has checksum 00000000

gardners, 2023-10-15:
Attempting to simulate the MEGA65 retro-computer, I am hitting a problem where I get the following error from the following command:

```
nvc -M 256m -L . -a --relaxed src/vhdl/shadowram-a100t.vhdl src/vhdl/debugtools.vhdl src/vhdl/crc.vhdl src/vhdl/victypes.vhdl src/vhdl/cputypes.vhdl  src/vhdl/gen_utils.vhdl src/vhdl/conversions.vhdl src/vhdl/s27kl0641-pgs-modified.vhd src/vhdl/fake_expansion_port.vhdl src/vhdl/fake_sdcard.vhdl src/vhdl/fake_reconfig.vhdl src/vhdl/fake_opl2.vhdl src/vhdl/dummy_uart_monitor.vhdl  src/vhdl/ddrwrapper.vhdl src/vhdl/hyppo.vhdl src/vhdl/version.vhdl src/vhdl/ghdl_chipram8bit.vhdl src/vhdl/ghdl_farstack.vhdl src/vhdl/colourram.vhdl src/vhdl/charrom.vhdl src/vhdl/ghdl_ram128x1k.vhdl src/vhdl/ghdl_ram32x1024.vhdl src/vhdl/ghdl_ram18x2k.vhdl src/vhdl/ghdl_ram8x4096.vhdl src/vhdl/ghdl_ram8x4096_sync.vhdl src/vhdl/ghdl_ram32x1024_sync.vhdl src/vhdl/ghdl_ram8x512.vhdl src/vhdl/ghdl_ram9x4k.vhdl src/vhdl/ghdl_screen_ram_buffer.vhdl src/vhdl/ghdl_videobuffer.vhdl src/vhdl/ghdl_ram36x1k.vhdl src/vhdl/asym_ram.vhdl src/vhdl/shadowram-a100t.vhdl src/vhdl/shadowram-a200t.vhdl src/vhdl/frame_generator.vhdl src/vhdl/pixel_driver.vhdl src/vhdl/pixel_fifo.vhdl src/vhdl/sprite.vhdl src/vhdl/bitplane.vhdl src/vhdl/bitplanes.vhdl src/vhdl/vicii_sprites.vhdl src/vhdl/pal_simulation.vhdl src/vhdl/ghdl_alpha_blend.vhdl src/vhdl/lfsr16.vhdl src/vhdl/oskmem.vhdl src/vhdl/visual_keyboard.vhdl src/vhdl/uart_charrom.vhdl src/vhdl/termmem.vhdl src/vhdl/rain.vhdl  src/vhdl/viciv.vhdl  src/vhdl/i2c_master.vhdl src/vhdl/i2c_wrapper.vhdl src/vhdl/i2c_controller.vhdl src/vhdl/grove_i2c.vhdl src/vhdl/hdmi_i2c.vhdl src/vhdl/spdif_encoder.vhdl src/vhdl/hdmi_spdif.vhdl src/vhdl/UART_TX_CTRL.vhdl src/vhdl/uart_rx.vhdl src/vhdl/buffereduart.vhdl src/vhdl/touch.vhdl src/vhdl/hyperram.vhdl src/vhdl/crc1581.vhdl src/vhdl/mfm_bits_to_bytes.vhdl src/vhdl/mfm_bits_to_gaps.vhdl src/vhdl/mfm_gaps_to_bits.vhdl src/vhdl/mfm_gaps.vhdl src/vhdl/mfm_quantise_gaps.vhdl src/vhdl/rll27_quantise_gaps.vhdl src/vhdl/rll27_bits_to_gaps.vhdl src/vhdl/rll27_gaps_to_bits.vhdl src/vhdl/raw_bits_to_gaps.vhdl src/vhdl/mfm_decoder.vhdl src/vhdl/ghdl_ram8x2048.vhdl src/vhdl/ethernet_miim.vhdl src/vhdl/ethernet.vhdl src/vhdl/ghdl_fpgatemp.vhdl src/vhdl/expansion_port_controller.vhdl src/vhdl/slow_devices.vhdl src/vhdl/sdcardio.vhdl src/vhdl/audio_mixer.vhdl src/vhdl/pdm_to_pcm.vhdl src/vhdl/pcm_to_pdm.vhdl src/vhdl/i2s_clock.vhdl src/vhdl/i2s_transceiver.vhdl src/vhdl/pcm_clock.vhdl src/vhdl/pcm_transceiver.vhdl src/vhdl/audio_complex.vhdl  src/vhdl/accessible_keyboard.vhdl src/vhdl/kb_matrix_ram.vhdl src/vhdl/keyboard_to_matrix.vhdl src/vhdl/matrix_to_ascii.vhdl src/vhdl/widget_to_matrix.vhdl src/vhdl/ps2_to_matrix.vhdl src/vhdl/keymapper.vhdl src/vhdl/virtual_to_matrix.vhdl src/vhdl/keyboard_complex.vhdl  src/vhdl/driverom.vhdl src/vhdl/dpram8x4096.vhdl src/vhdl/dummy_cpu6502.vhdl src/vhdl/internal1541.vhdl  src/vhdl/ps2_to_uart.vhdl src/vhdl/dummy_uart_monitor.vhdl  src/vhdl/vfpga/overlay_IP.vhdl src/vhdl/vfpga/vfpga_clock_controller_pausable.vhdl src/vhdl/vfpga/vfpga_wrapper_8bit.vhdl src/vhdl/framepacker.vhdl src/vhdl/mega65r2_i2c.vhdl src/vhdl/mega65r3_i2c.vhdl src/vhdl/mega65r4_i2c.vhdl src/vhdl/edid_i2c.vhdl src/vhdl/sid_coeffs.vhdl src/vhdl/sid_coeffs_mux.vhdl src/vhdl/sid_components.vhdl src/vhdl/sid_filters.vhdl src/vhdl/sid_voice.vhdl src/vhdl/sid_tables.vhdl src/vhdl/ghdl_8580_voice_stub.vhdl src/vhdl/sid_6581.vhdl  src/vhdl/mouse_input.vhdl src/vhdl/cia6526.vhdl src/vhdl/c65uart.vhdl src/vhdl/iomapper.vhdl  src/vhdl/multiply32.vhdl src/vhdl/divider32.vhdl src/vhdl/fast_divide.vhdl src/vhdl/shifter32.vhdl src/vhdl/neotrng.vhdl src/vhdl/gs4510.vhdl src/vhdl/machine.vhdl  src/vhdl/cpu_test.vhdl src/vhdl/debugtools.vhdl src/vhdl/fake_expansion_port.vhdl -e cpu_test -r
** Warning: shared variable RAM must have protected type
        > src/vhdl/shadowram-a100t.vhdl:393243
        |
 393243 |   shared variable ram : ram_t := initram;
        |                   ^^^
** Fatal: /home/paul/Projects/mega65/mega65-core/work/WORK.FAKE_EXPANSION_PORT: design unit depends on WORK.DEBUGTOOLS with checksum a4cadebc 
          but the current version in the library has checksum 00000000
** Note: this usually means WORK.FAKE_EXPANSION_PORT is outdated and needs to be reanalysed
make: *** [Makefile:479: simulate-nvc] Error 1
```

Note that debugtools.vhdl is in the file list ahead of fake_expansion_port.vhdl

Adding both files a 2nd time to the end as I have done above doesn't change the result.  

I can provide a tarball of the files if helpful, or if you're keen, its all in the pgs-r4-fixups branch of github.com/mega65/mega65-core, but it does have some odd dependencies, thus my offer to make you a tarball to help investigate it.

nickg, 2023-10-15:
I ran into some errors trying to build `src/vhdl/shadowram-a100t.vhdl` (I was using a Mac though). I'll have another go on my Linux machine this evening. My email is in my GitHub profile if you want to send the tarball (best to send a link to Google Drive or something).

gardners, 2023-10-15:
Email with tarball sent.

commit 3f718447 (2023-10-15): Fix cyclic dependency bug with "use work.all"

Fixes #769
commit ba5298a8 (2023-10-16): Fix cyclic dependency bug with "use work.all"

Fixes #769
closed 2023-10-15 by nickg