nickg/nvc #770: Problem with mutiple warnings and --relaxed option

Anselmo95, 2023-10-15:
Hello,

When using the option `--relaxed` the analisys fails if a file contains two or more warnings of the same type.
For example this code:
```
library ieee;
use ieee.std_logic_1164.all;

package pkg is
    function identity_a (a : std_logic) return std_logic;    
    function identity_b (b : std_logic) return std_logic;    
end package pkg;

package body pkg is
    function identity_a (constant a : std_logic) return std_logic is
    begin
        return a; 
    end function;
    
    function identity_b (constant b : std_logic) return std_logic is
    begin
        return b; 
    end function;
end package body pkg;
```

The problem seams to be inside the `sem.c` file, specifically the macro/function combo `pedantic_diag/_pedantic_diag` which uses the static variable `static int _warned = 0;`. On the first macro call everything is fine but on the second one `_warned` will be at 1, having keept the value from the prevuois call, and this leads to the return of a null pointer which is later dereferenced when tring to print the warning message.

As a dumb fix just remowing the `static` keyword fix the issue but maybe there is a reason why it is there

nickg, 2023-10-15:
The result of `pedantic_diag()` is supposed to be checked for NULL. I fixed a few other places this could happen.

closed 2023-10-15 by nickg

Anselmo95, 2023-10-16:
Thanks for the quick fix as always!

But won't this hide warning messages?
What I mean is that for the example above there will be a warning for `identity_a` but nothing for `identity_b` even if both have the same structure

nickg, 2023-10-16:
> 
> But won't this hide warning messages? What I mean is that for the example above there will be a warning for `identity_a` but nothing for `identity_b` even if both have the same structure

Yes that's the intended behaviour: it prints each warning once and then silences it. Otherwise with `--relaxed` you might get many warnings for the same thing which would be annoying.