nickg/nvc #775: Can't start simulation with latest devel: `(init) missing body for`

m42uko, 2023-10-20:
After updating from 1.10.4 to 1.11-devel (1.10.0.r255.g15a77fe1), I can't start a simulation anymore. It errors out with `** Fatal: (init): missing body for WORK.DFF_TEST_VHDL`. I'm running on ArchLinux.

I noticed this in testing for https://github.com/themperek/cocotb-test/pull/248, but it also happens when trying to compile and start the simulation manually. See below for a full log.

Do I need to do something differently in the new version?  Or did something change in how NVC must be built?

```
[builduser@737d440cb9bd my_workspace]$ nvc -a /home/builduser/cocotb-test/src/cocotb-test/tests/dff.vhdl
[builduser@737d440cb9bd my_workspace]$ nvc -e dff_test_vhdl
[builduser@737d440cb9bd my_workspace]$ nvc -r dff_test_vhdl
** Fatal: (init): missing body for WORK.DFF_TEST_VHDL
[builduser@737d440cb9bd my_workspace]$ cat /home/builduser/cocotb-test/src/cocotb-test/tests/dff.vhdl
-- This file is public domain, it can be freely copied without restrictions.
-- SPDX-License-Identifier: CC0-1.0

library ieee;
use ieee.std_logic_1164.all;

entity dff_test_vhdl is
port(
  clk: in std_logic;
  d: in std_logic;
  q: out std_logic);
end dff_test_vhdl;

architecture behavioral of dff_test_vhdl is
begin
  process (clk) begin
    if rising_edge(clk) then
      q <= d;
    end if;
  end process;
end behavioral;
[builduser@737d440cb9bd my_workspace]$ nvc -v
nvc 1.11-devel (1.10.0.r255.g15a77fe1) (Using LLVM 16.0.6)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

nickg, 2023-10-20:
It's working ok for me. How did you configure/build?

m42uko, 2023-10-20:
Oh, interesting. I'm building according to this ArchLinux PKGBUILD: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=nvc-git

The relevant steps are in `build()`.

nickg, 2023-10-20:
Remove the `--enable-jit` option. That was a just an experimental thing for testing, I should remove it from the configure script.

m42uko, 2023-10-20:
Thanks, that was indeed the issue. No clue why that was there in the first place.

nickg, 2023-10-20:
I removed the option completely in 4cf96f6

closed 2023-10-20 by nickg