nickg/nvc #778: Minor: whitespace in work library command line argument

amb5l, 2023-10-23:
Imperfections in my makefiles sometimes result in leading and/or trailing whitespace in library names.

Running
`nvc --work= work -a filename.vhd`
produces an error:
`** Fatal: invalid library name`

Please could nvc strip leading and trailing whitespace from the library name?

nickg, 2023-10-23:
The work library name here ends up being "" rather than " work" because the getopt routine we used for option parsing parses `--work= work` as a named option `--work` with empty argument and a separate positional argument `work`. I've modified the error message a bit so it's more clear and reports "library name cannot be empty instead".

In the make file I think you could use `--work $(NAME)` instead (i.e. without the `=`). That should be more forgiving of extra whitespace as it will be eaten by the shell.

amb5l, 2023-10-23:
Fair enough and please close the issue. Slight error in my report - the space crept in because of some poor formatting of lists in a TCL script. I'm trimming strings that are passed to --work now.

closed 2023-10-23 by nickg