nickg/nvc #781: Add VITAL95 libraries

Blebowski, 2023-10-24:
Currently, NVC compilation includes only VITAL 2000 libraries. Some older devices still may use
VITAL95 libraries. I am unsure about the backwards compatibility. 
For SDF annotation, it may be good if NVC provided both VITAL versions.

Switching between VITAL versions could be done by a switch (`--vital-std=<95|2000>`) ?
By default, maybe 2000.

The libraries can be found online (e.g. GHDL repo has them).

I am not sure what is the good way to implement this. Since the Vital version is orthogonal
to VHDL version, the compiled library would need to be split from `preload.X`. However, since
mixing multiple VHDL versions is not possible, it would most likely end-up in having 6 different
VITAL library instances for each combination of VHDL 93,2008,2019 and VITAL 95,2000.

nickg, 2023-10-24:
I diff-ed the 95/2000 libraries and the changes seem quite minimal: either adding extra optional parameters or changing a `signal` parameters to `constant`. The other changes that don't affect the interfaces seem to be bug fixes. So I don't think there will be any compatibility problems with just providing the 2000 libraries.

Blebowski, 2023-10-25:
Ok, so then lets keep the 2000 only.

closed 2023-10-25 by Blebowski