nickg/nvc #782: EXCEPTION_ACCESS_VIOLATION

avelure, 2023-10-25:
I get this in a big project I haven't tested in a while with NVC.
From the UVVM printout at the start it seems it started simulating.

```
*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

*** Caught exception c0000005 (EXCEPTION_ACCESS_VIOLATION) [address=00000000000000E8, ip=00007FF7B61672E7] ***

[00007FF7B6089FFF]
[00007FF7B608AB5C]
[00007FFD7D39E237] UnhandledExceptionFilter+0x1e7
[00007FFD7CD4F0B8]
[00007FFD7CD4EEA1]
[00007FFD7FB354B0] memset+0x13b0
[00007FFD7FB1CA06] _C_specific_handler+0x96
[00007FFD7FB323AF] _chkstk+0x11f
[00007FFD7FAE14B4] RtlRaiseException+0x484
[00007FFD7FB30EBE] KiUserExceptionDispatcher+0x2e
[00007FF7B61672E7] _nvc_test_event+0x3dc7
[00007FF7B6152B30] _nvc_sys_write+0x3fe0
[00007FF7B6154228] _nvc_sys_write+0x56d8
[00007FF7B614004F]
[00007FF7B60E523C]
[00007FF7B60F95B9]
[00007FF7B60F9628]
[00007FF7B60F94F7]
[00007FF7B60F9628]
[00007FF7B60F94F7]
[00007FF7B60F94F7]
[00007FF7B60F9628]
[00007FF7B60F94F7]
[00007FF7B60F94F7]
```

```

****************************************************************************************************
*
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
****************************************************************************************************
*

[New Thread 32284.0x36fc]
[New Thread 32284.0x1fe8]
[New Thread 32284.0xb508]
[New Thread 32284.0xac98]
[New Thread 32284.0x9cec]
[New Thread 32284.0x24a4]
[New Thread 32284.0x6e8c]

Thread 1 received signal SIGSEGV, Segmentation fault.
jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>, tlab=<optimized out>)
    at ../src/jit/jit-interp.c:503
503           state->regs[ir->result].integer = *(int64_t *)arg1;
(gdb) bt
#0  jit_interp (f=<optimized out>, caller=<optimized out>, args=<optimized out>,
    tlab=<optimized out>) at ../src/jit/jit-interp.c:503
#1  0x00007ff7b6152b30 in jit_try_vcall.lto_priv.0 (j=j@entry=0x7432f0, f=f@entry=0xaf2ace0,
    result=result@entry=0x5fe9f8, args=args@entry=0x5fea00) at ../src/jit/jit-core.c:681
#2  0x00007ff7b6154228 in jit_call_thunk (j=0x7432f0, unit=<optimized out>,
    context=<optimized out>, fn=0x7ff7b61385d0 <thunk_result_cb>, arg=0x47a5c10)
    at ../src/jit/jit-core.c:822
#3  0x00007ff7b614004f in eval_do_fold.lto_priv.0 (jit=0x7432f0, expr=0x47a5c10,
    parent=<optimized out>, context=0x0) at ../src/eval.c:186
#4  0x00007ff7b60e523c in simp_tree.lto_priv.0 (t=0x47a5c10, _ctx=0x5ff440) at ../src/eval.c:219
#5  0x00007ff7b60f95b9 in object_rewrite (object=object@entry=0x47a5c10, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:675
#6  0x00007ff7b60f9628 in object_rewrite (object=0x47a5c50, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:738
#7  0x00007ff7b60f94f7 in object_rewrite (object=object@entry=0x47a5bd0, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:749
#8  0x00007ff7b60f9628 in object_rewrite (object=0x47a5c80, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:738
#9  0x00007ff7b60f94f7 in object_rewrite (object=0x47a5cc0, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:749
#10 0x00007ff7b60f94f7 in object_rewrite (object=object@entry=0x47a5cf0, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:749
#11 0x00007ff7b60f9628 in object_rewrite (object=0x47a5d30, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:738
#12 0x00007ff7b60f94f7 in object_rewrite (object=0x47a6320, ctx=ctx@entry=0x5ff3a0)
    at ../src/object.c:749
#13 0x00007ff7b60f94f7 in object_rewrite (object=0x47a6370, ctx=0x5ff3a0) at ../src/object.c:749
#14 0x00007ff7b619a22f in tree_rewrite.isra.0 () at ../src/tree.c:1283
#15 0x00007ff7b60e205b in simplify_global (top=<optimized out>, generics=0x0, jit=<optimized out>,
    ur=<optimized out>) at ../src/simp.c:1553
#16 0x00007ff7b60e3c19 in elab_stmts.lto_priv.0 (t=0x47492e0, ctx=<optimized out>)
    at ../src/elab.c:1475
#17 0x00007ff7b60e3c97 in elab_stmts.lto_priv.0 (t=0x4748b70, ctx=<optimized out>)
    at ../src/elab.c:1488
#18 0x00007ff7b60e3c97 in elab_stmts.lto_priv.0 (t=t@entry=0x47426a0, ctx=<optimized out>,
    ctx@entry=0x5ffa20) at ../src/elab.c:1488
#19 0x00007ff7b60e4f7c in elab_top_level.lto_priv.0 (arch=<optimized out>, ename=<optimized out>,
    ctx=0x5ffb70) at ../src/elab.c:2108
#20 0x00007ff7b6088b96 in elaborate (argc=argc@entry=12, argv=<optimized out>,
    argv@entry=0x6ac6f0, state=state@entry=0x5ffe30) at ../src/tree.c:550
#21 0x00007ff7b6085b67 in process_command (argc=argc@entry=12, argv=argv@entry=0x6ac6f0,
    state=state@entry=0x5ffe30) at ../src/nvc.c:1817
#22 0x00007ff7b61af64d in main (argc=<optimized out>, argv=<optimized out>) at ../src/nvc.c:1950
```
This was on Windows, I get something similar on Linux.
```
*** Caught signal 11 (SEGV_MAPERR) [address=0xe8, ip=0x564bcf2c92fd] ***
[0x564bcf1e544d] ../src/util.c:860 signal_handler.lto_priv.0
[0x7f8f198b551f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x564bcf2c92fd] ../src/jit/jit-interp.c:503 interp_loop
[0x564bcf2c92fd] ../src/jit/jit-interp.c:[108](https://gitlab.inventas.no/kongsberg-space/elektra-fgp/-/jobs/67002#L108)9 jit_interp
[0x564bcf2b4c07] ../src/jit/jit-core.c:681 jit_try_vcall.lto_priv.0
[0x564bcf2b4e3b] ../src/jit/jit-core.c:822 jit_call_thunk
[0x564bcf29904b] ../src/eval.c:186 eval_do_fold.lto_priv.0
[0x564bcf243310] ../src/simp.c:219 simp_type_conv
[0x564bcf243310] ../src/eval.c:1436 simp_tree.lto_priv.0
[0x564bcf2583ec] ../src/object.c:675 object_rewrite
[0x564bcf2584ff] ../src/object.c:738 object_rewrite
[0x564bcf25831c] ../src/object.c:749 object_rewrite
[0x564bcf2584ff] ../src/object.c:738 object_rewrite
[0x564bcf25831c] ../src/object.c:749 object_rewrite
[0x564bcf25831c] ../src/object.c:749 object_rewrite
[0x564bcf2584ff] ../src/object.c:738 object_rewrite
[0x564bcf25831c] ../src/object.c:749 object_rewrite
[0x564bcf25831c] ../src/object.c:749 object_rewrite
[0x564bcf2ff70f] ../src/tree.c:1283 tree_rewrite.isra.0
[0x564bcf23fe1f] ../src/simp.c:1553 simplify_global
[0x564bcf241c64] ../src/elab.c:1475 elab_stmts.lto_priv.0
[0x564bcf241dfe] ../src/elab.c:1488 elab_stmts.lto_priv.0
[0x564bcf241dfe] ../src/elab.c:1488 elab_stmts.lto_priv.0
[0x564bcf242fd5] ../src/elab.c:2108 elab_top_level.lto_priv.0
[0x564bcf1ddda5] ../src/elab.c:2158 elab
[0x564bcf1ddda5] ../src/elab.c:456 elaborate
[0x564bcf1deb3f] ../src/nvc.c:1817 process_command
[0x564bcf1da0c0] ../src/nvc.c:1950 main
Error: Program ended with exit code 2
```

nickg, 2023-10-26:
It's caused by a type conversion. Can you try making the change below and re-run? It should print the location just before it crashes.

```diff
diff --git a/src/simp.c b/src/simp.c
index 9414c59c5cd1..198649c61219 100644
--- a/src/simp.c
+++ b/src/simp.c
@@ -359,8 +359,10 @@ static tree_t simp_type_conv(tree_t t, simp_ctx_t *ctx)
          return t;   // Not supported currently
    }
 
-   if (eval_possible(t, ctx->registry))
+   if (eval_possible(t, ctx->registry)) {
+      fmt_loc(stdout, tree_loc(t));
       return eval_try_fold(ctx->jit, t, NULL, NULL);
+   }
 
    return t;
 }
```

avelure, 2023-10-26:
It does print the location twice, but if I do an MWE it does not crash

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  generic (
    GC_SAMPLES_PER_BIT             : natural
  );
end entity test;
architecture beh of test is
begin
  
  b_tx_status_timer : block is
    constant C_MAX_CYCLES_NEW_WORD      : natural := natural(5.5 * real(GC_SAMPLES_PER_BIT));
  begin
  process
  begin
    report "OK";
    wait;
  end process;
  end block;

end architecture beh;
```
GC_SAMPLES_PER_BIT is 16. It is a constant passed through the hierarchy, but not from the top level.

nickg, 2023-10-26:
Thanks. It's a bit mysterious. Could you also try setting the environment variable `export NVC_LOWER_VERBOSE=1` and then re-running. It should dump lots of output that looks like:

```
                                                                                               
Kind       thunk                                                                               
Blocks     1                                                                                   
Registers  6                                                                                   
Types      5                                                                                   
Variables  0                                                                                   
Result     -2^63..2^63-1                                                                       
Begin                                                                                          
   0: r0 := const 1                     // -2^63..2^63-1 => 1                                  
      r1 := const 0                     // -2^63..2^63-1 => 0                                  
      r2 := debug locus WORK.TRACE1-body-35 // D<>                                             
      zero check r1 locus r2                                                                   
      r3 := div r0 / r1                 // -2^63..2^63-1                                       
      r4 := cast r3                     // -2^31..2^31-1 => -2^63..2^63-1                      
      r5 := cast r4                     // -2^63..2^63-1
      return r5
```

Can you paste the last one just before it crashes?

avelure, 2023-10-26:
This is the last block before it crashes, however it is not in this package that the file location was printed
```
Name       DESIGN_LIB.EEPROM_PKG
Kind       package
Blocks     35
Registers  799
Types      76
  DESIGN_LIB.EEPROM_PKG.T_EEPROM_OUTPUT {0..8, 0..8, 0..2}
  DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT  {[24] : 0..8, [*] : 0..8, 0..15, -2^31..2^31-1, 0..1}
Variables  51
  C_POWER_STABLE_DELAY                  // -2^31..2^31-1 => 0..2^31-1, constant
  C_IMAGE_CHECK_VALUE                   // [16] : 0..8 => 0..8, constant
  C_NUM_CRC_IN_IMAGE                    // -2^31..2^31-1 => 0..2^31-1, constant
  C_DATA_INVALID_VERSION                // [8] : 0..8 => 0..8, constant
  C_CAL_TABLE_SIZE                      // -2^31..2^31-1 => 0..2^31-1, constant
  C_CAL_CRC_POLYNOMIAL                  // [16] : 0..8 => 0..8, constant
  C_CAL_CRC_INIT                        // [16] : 0..8 => 0..8, constant
  C_CRC_OK                              // [16] : 0..8 => 0..8, constant
  C_IMAGE_CHECK                         // -2^31..2^31-1 => 0..2^31-1, constant
  C_INTERVAL_BETWEEN_COMMANDS           // -2^31..2^31-1 => 0..2^31-1, constant
  C_NUM_INTERMEDIATE_POINTS             // -2^31..2^31-1 => 0..2^31-1, constant
  C_ACC_LIMIT_DEFAULT                   // -2^31..2^31-1 => 0..2^31-1, constant
  C_VEL_LIMIT_DEFAULT                   // -2^31..2^31-1 => 0..2^31-1, constant
  C_POWER                               // -2^31..2^31-1 => 0..2^31-1, constant
  C_STEP_MODE                           // -2^31..2^31-1 => 0..2^31-1, constant
  C_MIL1553_THRESHOLD_1_SYNC            // -2^31..2^31-1 => 0..2^31-1, constant
  C_MIL1553_THRESHOLD_2_DATA            // -2^31..2^31-1 => 0..2^31-1, constant
  C_MIL1553_THRESHOLD_3_NO_BIT          // -2^31..2^31-1 => 0..2^31-1, constant
  C_PARAMETERS_CRC                      // -2^31..2^31-1 => 0..2^31-1, constant
  C_SINE_WAVE                           // -2^31..2^31-1 => 0..2^31-1, constant
  C_SINE_WAVE_CRC                       // -2^31..2^31-1 => 0..2^31-1, constant
  C_VERSION_NUMBER                      // -2^31..2^31-1 => 0..2^31-1, constant
  C_CAL_A                               // -2^31..2^31-1 => 0..2^31-1, constant
  C_CAL_A_CRC                           // -2^31..2^31-1 => 0..2^31-1, constant
  C_CAL_B                               // -2^31..2^31-1 => 0..2^31-1, constant
  C_CAL_B_CRC                           // -2^31..2^31-1 => 0..2^31-1, constant
  C_EEPROM_IMAGE_SIZE                   // -2^31..2^31-1 => 0..2^31-1, constant
  C_EEPROM_IMAGE_2_START                // -2^31..2^31-1 => 0..2^31-1, constant
  C_IMAGE_SEARCH_TIME_OUT               // -2^31..2^31-1 => 0..2^31-1, constant
  C_EEPROM_BYTE_CNT_WIDTH               // -2^31..2^31-1 => 0..2^31-1, constant
  C_EERPOM_POWER_STATE                  // [5] : DESIGN_LIB.EEPROM_PKG.T_EEPROM_OUTPUT{} => DESIGN_LIB.EEPROM_PKG.T_EEPROM_OUTPUT{}, constant
  C_MAX_WORD_LENGTH                     // -2^31..2^31-1 => 0..2^31-1, constant
  C_MAX_ADDR_WIDTH                      // -2^31..2^31-1 => 0..2^31-1, constant
  DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT_TEMPLATE // [*] : DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{} => DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, constant
  C_SEGMENT_RECORDS                     // [*] : DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{} => DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, constant
  record139                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record140                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record141                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record142                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record143                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record144                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record145                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record146                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record147                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record148                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record149                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record150                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record151                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record152                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record153                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
  record154                             // DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}, temp
Begin
   0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
      r1 := package init IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
      r2 := package init IEEE.NUMERIC_STD // P<IEEE.NUMERIC_STD>
      r3 := package init DESIGN_LIB.TOP_PKG // P<DESIGN_LIB.TOP_PKG>
      r4 := const 4095                  // -2^31..2^31-1 => 4095
      r5 := const 0                     // -2^31..2^31-1 => 0
      r7 := const 0                     // 0..1 => 0
      C_POWER_STABLE_DELAY := store r4
[... 843 lines cut ...]
      r772 := store indirect r777
      r778 := record ref r769 field 2   // @<0..15> => 0..15
      r778 := store indirect r767
      r779 := record ref r769 field 3   // @<-2^31..2^31-1> => -2^31..2^31-1
      r779 := store indirect r768
[... cut ...]
      r789 := array ref r103 offset r788 // @<DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}> => DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}
      r789 := copy r769
      r790 := unwrap r104               // @<DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}> => DESIGN_LIB.EEPROM_PKG.T_DATA_SEGMENT{}
      r794 := uarray right r104 dim 0   // #
      r795 := cast r794                 // 0..15
      r796 := range null left r783 right r795 dir r786 // 0..1
      cond r796 then 34 else 33
  33: jump 34
  34: r103 := copy r790 count r8
      return
```

nickg, 2023-10-27:
Is the value of `GC_SAMPLES_PER_BIT` derived from that package in some way?

avelure, 2023-10-27:
No, it is a literal constant set in a generic two levels up.
I now realized I have a module testbench of this block (the crash happens running on top level), but running the module test does not crate a crash. The generic value, code and the debug printouts are the same.

But I now see that the traceback is different on windows and linux. If I add `fmt_loc(stdout, tree_loc(expr));` in `eval_try_fold` I instead get another last one that is in that package.
When recreating that one I get the error.

```vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

package top_pkg is
  function log2_ceil (
    val: natural)
  return natural;
end package top_pkg;

package body top_pkg is

  function log2_ceil (
    val: natural)
  return natural is
  begin
    return integer(ceil(log2(real(val) + 1.0)));
  end function log2_ceil;
end package body top_pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.top_pkg.log2_ceil;

package test_pkg is
  type t_segment_type is (
    VERSION_NUMBER
  );
  constant C_MAX_WORD_LENGTH : natural := 3;
  constant C_MAX_ADDR_WIDTH  : natural := log2_ceil(4095);
  type t_data_segment is record
    word_idx     : unsigned(C_MAX_ADDR_WIDTH - 1 downto 0);
    word_length  : natural range 0 to C_MAX_WORD_LENGTH;
  end record t_data_segment;

  type     t_data_segment_template is array(t_segment_type) of t_data_segment;
  constant C_SEGMENT_RECORDS : t_data_segment_template := (
    VERSION_NUMBER            => t_data_segment'(
      word_idx                => (others => '0'),
      word_length             => 1)
  );

end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.test_pkg.all;

entity test is
end entity test;
architecture beh of test is
  signal segment_word        : t_data_segment;
begin
  
  b_block : block is
   signal calib_version_tmp : std_logic_vector(C_SEGMENT_RECORDS(VERSION_NUMBER).word_length * 8 - 1 downto 0);
  begin
  process
  begin
    report "OK 2";
    wait;
  end process;
  end block;

end architecture beh;
```

avelure, 2023-10-27:
I can confirm that if I add `fmt_loc(stdout, tree_loc(expr)); in eval_do_fold :187 on linux , the last line that is indicated is the same as on windows when I run thw original testbench, which is the line that is recreated in the last MWE.
So there is just a difference in the traceback.

nickg, 2023-10-27:
That's great, thank you very much for providing the small reproducer! I think this should be fixed now, could you test again?

avelure, 2023-10-28:
Thanks for the quick fix, it works now on both linux and windows.

closed 2023-10-28 by nickg