nickg/nvc #786: Windows msys2 clang64 compile warnings

avelure, 2023-11-01:
I noticed a few warning compiling with updated Msys2 Clang64 on windows, however they don't seem very critical.

```
src/lexer.c:1807:13: warning: misleading indentation; statement is not part of the previous 'if'
      [-Wmisleading-indentation]
 1807 |             if ( ! (yy_state_buf) )
      |             ^
src/lexer.c:1805:9: note: previous statement is here
 1805 |         if ( ! (yy_state_buf) )
      |         ^
src/lexer.c:3585:3: warning: misleading indentation; statement is not part of the previous 'if'
      [-Wmisleading-indentation]
 3585 |                 return yy_is_jam ? 0 : yy_current_state;
      |                 ^
src/lexer.c:3582:2: note: previous statement is here
 3582 |         if ( ! yy_is_jam )
      |         ^
2 warnings generated.
---- snip -----
../src/thread.c:800:11: warning: variable 'comp' set but not used [-Wunused-but-set-variable]
  800 |       int comp = 0;
      |           ^
1 warning generated.
---- snip -----
../src/jit/jit-exits.c:427:6: warning: redeclaration of '__nvc_do_exit' should not add 'dllexport'
      attribute [-Wdll-attribute-on-redeclaration]
  427 | void __nvc_do_exit(jit_exit_t which, jit_anchor_t *anchor, jit_scalar_t *args,
      |      ^
../src/jit/jit-priv.h:423:6: note: previous declaration is here
  423 | void __nvc_do_exit(jit_exit_t which, jit_anchor_t *anchor, jit_scalar_t *args,
      |      ^
../src/jit/jit-exits.c:1013:6: warning: redeclaration of '__nvc_do_fficall' should not add
      'dllexport' attribute [-Wdll-attribute-on-redeclaration]
 1013 | void __nvc_do_fficall(jit_foreign_t *ff, jit_anchor_t *anchor,
      |      ^
../src/jit/jit-priv.h:425:6: note: previous declaration is here
  425 | void __nvc_do_fficall(jit_foreign_t *ff, jit_anchor_t *anchor,
      |      ^
../src/jit/jit-exits.c:1027:6: warning: redeclaration of '_debug_out' should not add 'dllexport'
      attribute [-Wdll-attribute-on-redeclaration]
 1027 | void _debug_out(intptr_t val, int32_t reg)
      |      ^
../src/jit/jit-priv.h:428:6: note: previous declaration is here
  428 | void _debug_out(intptr_t val, int32_t reg);
      |      ^
../src/jit/jit-exits.c:1049:7: warning: redeclaration of '__nvc_mspace_alloc' should not add
      'dllexport' attribute [-Wdll-attribute-on-redeclaration]
 1049 | void *__nvc_mspace_alloc(uintptr_t size, jit_anchor_t *anchor)
      |       ^
../src/jit/jit-priv.h:427:7: note: previous declaration is here
  427 | void *__nvc_mspace_alloc(uintptr_t size, jit_anchor_t *anchor);
      |       ^
4 warnings generated.
---- snip -----
../thirdparty/fstapi.c:6033:14: warning: variable 'secnum' set but not used
      [-Wunused-but-set-variable]
 6033 | unsigned int secnum = 0;
      |              ^
1 warning generated.
---- snip -----
  CCLD     bin/nvc.exe
gcc: warning: argument unused during compilation: '-rdynamic' [-Wunused-command-line-argument]

```

avelure, 2023-11-07:
A few more from latest master with gui
```
../src/server.c:204:25: warning: passing 'uint8_t *' (aka 'unsigned char *') to parameter of type
      'const char *' converts between pointers to integer types where one is of the unique plain
      'char' type and the other is not [-Wpointer-sign]
  204 |          send(ws->sock, ws->tx_buf + ws->tx_rptr, chunksz, 0);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/clang64/include/winsock2.h:1033:60: note: passing argument to parameter 'buf' here
 1033 |   WINSOCK_API_LINKAGE int WSAAPI send(SOCKET s,const char *buf,int len,int flags);
      |                                                            ^
../src/server.c:255:18: warning: unused variable 'fin' [-Wunused-variable]
  255 |       const bool fin = frame[0] & 0x80;
      |                  ^~~
2 warnings generated.

```
I also had to turn off the lint check for the JS files as my git checks out text files with windows endings automatically
```
C:\proj\public\nvc\contrib\gui\.eslintrc.js
   1:19  error  Expected linebreaks to be 'LF' but found 'CRLF'  linebreak-style
```

nickg, 2023-11-07:
I've fixed some of these and turned off that lint check.

nickg, 2024-07-10:
All the warnings on the clang64 build are fixed now except the "misleading indentation" one. This comes from some code generated by Flex so not much I can do about that. I see the same warning on macOS and FreeBSD which also use Clang.

closed 2024-07-10 by nickg