nickg/nvc #787: Negative/Null range

Anselmo95, 2023-11-01:
Hello,

Here is some code that makes nvc crash during simulation and which I believe is linked to the use of negative or null ranges.
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent2 is
generic (
    M  : natural := 10;
    W  : natural := 0
);
port (
    i : in  std_logic_vector(M-1 downto 0);
    o : out std_logic_vector(M+W-1 downto 0)
);
end entity ent2;

architecture rtl of ent2 is
    signal s0 : std_logic_vector(W-1 downto 0);
begin
    s0 <= (others => 'X');
    o <= i & s0;
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent1 is
end entity ent1;

architecture rtl of ent1 is
    constant M : natural := 7;
    constant W : natural := 0;
    signal i : std_logic_vector(M-1 downto 0);
    signal o : std_logic_vector(M+W-1 downto 0);
    signal o_l : std_logic_vector(W-1 downto 0);
    signal o_h : std_logic_vector(M-1 downto 0);
begin

    process
    begin
        i <= (others => '0');
        wait;
    end process;

    ent2_inst : entity work.ent2
    generic map (
        M => M,
        W => W
    )
    port map (
        i => i,
        o => o
    );

    o_l <= o(W-1 downto 0);
    o_h <= o(M+W-1 downto W);

end architecture;
```
I think the problem is probably inside [model.c](https://github.com/nickg/nvc/blob/06516964b889429e37a7f311bee43ca8cb8306e4/src/rt/model.c#L3696), in the _x_transfer_signal_ function where the _t->source_ pointer for the _o_l_ signal is set to null causing the crash later on in the code.

I am not too sure how these kind of ranges should be treated, in case of vcs and questa they both print a warning but then run the simulation anyway, which is kind of surprising.

(NOTE: Sorry for the big reproducer code I couldn't make it smaller)

nickg, 2023-11-01:
> 
> I am not too sure how these kind of ranges should be treated, in case of vcs and questa they both print a warning but then run the simulation anyway, which is kind of surprising.
> 

This code is actually fine: if the length is negative it is treated as zero. I guess they give a warning because usually a null array is not intended.

closed 2023-11-01 by nickg

Anselmo95, 2023-11-01:
> This code is actually fine: if the length is negative it is treated as zero. I guess they give a warning because usually a null array is not intended.

This makes sense, thanks for the expenation and for the quick fix !

commit a16ee7f2 (2023-11-01): Fix crash with assignment to null signal. Fixes #787