nickg/nvc #789: Invalid use of architecture

avelure, 2023-11-02:
I see this testcase fails on all simulators, but I was wondering if it is supported by the LRM.
I guess it is down to ambiguity, but logically I would expect it to select the one that is part of the enum.
If I try to type qualify the value with `t_test'(RTL)`  it also fails in all simulators,

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type t_test is (GL, RTL);
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.test_pkg.all;

entity test1 is
  generic (
    GC_TEST_TYPE : t_test
  );
end entity;

architecture rtl of test1 is
begin
  g_test : if GC_TEST_TYPE = RTL generate
    p_proc : process 
    begin
      wait;
    end process;
  end generate g_test;
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.test_pkg.all;

entity test is
end entity;

architecture rtl of test is
  signal sig  : std_logic;
begin
  i_test1 : work.test1
  generic map (GC_TEST_TYPE => RTL);
end architecture;
```

```
** Error: invalid use of architecture RTL
    > C:\proj\public\minimal\test_generic_architecture.vhd:23
    |
 23 |   g_test : if GC_TEST_TYPE = RTL generate
    |                              ^^^
** Error: design unit WORK.TEST1 is not a component declaration
** Error: invalid use of architecture RTL
    > C:\proj\public\minimal\test_generic_architecture.vhd:44
    |
 44 |   generic map (GC_TEST_TYPE => RTL);
    |                                ^^^
```

But as you are interested in improving error messages, then maybe the message can be updated to make it more clear what the issue is.
GHDL reports 
```
test_generic_architecture.vhd:23:30: no declaration for "rtl" (due to conflicts)
  g_test : if GC_TEST_TYPE = RTL generate
```
which is not that much clearer, but mentions a conflict

Modelsim is even more incomprehensible
```
** Error: test_generic_architecture.vhd(23): (vcom-1590) Bad expression in right operand of infix expression '='.
** Error: test_generic_architecture.vhd(23): Type error resolving infix expression "=" as type std.STANDARD.BOOLEAN.
** Warning: test_generic_architecture.vhd(23): (vcom-1148) Condition in IF GENERATE must be static.
```

nickg, 2023-11-09:
Thanks for the test case. This now reports:

```
** Error: invalid use of architecture RTL
    > test.vhd:15
    |
  2 |   type t_test is (GL, RTL);
    |                       ^^^ declaration of literal RTL is hidden
 ...
 13 | architecture rtl of test1 is
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ name RTL refers to this architecture
 14 | begin
 15 |   g_test : if GC_TEST_TYPE = RTL generate
    |                              ^^^ error occurred here
** Error: design unit WORK.TEST1 is not a component declaration
    > test.vhd:31
    |
 31 |   i_test1 : work.test1
    |   ^^^^^^^^^^^^^^^^^^^^
** Error: invalid use of architecture RTL
    > test.vhd:32
    |
  2 |   type t_test is (GL, RTL);
    |                       ^^^ declaration of literal RTL is hidden
 ...
 28 | architecture rtl of test is
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^ name RTL refers to this architecture
 ...
 32 |   generic map (GC_TEST_TYPE => RTL);
    |                                ^^^ error occurred here
```

closed 2023-11-09 by nickg

avelure, 2023-11-10:
Thanks, this makes it a bit more clear.

commit ca26deb9 (2023-11-09): Improve error when architecture name hides object

Fixes #789