nickg/nvc #791: Stack overflow, regression

avelure, 2023-11-08:
With latest version I get a stack overflow. 
Last working version is a week ago 06516964b889429e37a7f311bee43ca8cb8306e4
GDB only gives me 
```
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff68e198a46 in ___chkstk_ms ()
(gdb) bt
#0  0x00007ff68e198a46 in ___chkstk_ms ()
#1  0x00000000001000d0 in ?? ()
```

This is without gui enabled in the configuration.
I'll try to do a bisect.

avelure, 2023-11-08:
I thought this was a regression, but it seems it is only present on CLANG64 (UCRT64 is ok).
The problem is when I declare a large 2D variable in a subprocess.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture beh of test is

  type t_slv_array is array (natural range <>) of std_logic_vector;

  procedure read_mem is
    variable v_memory        : t_slv_array(0 to 131072-1)(7 downto 0) := (others => x"FF");
  begin
  end procedure;
  
begin

  p_proc : process
  begin
    read_mem;
    wait;
  end process;
end architecture;
```

nickg, 2023-11-08:
I got a backtrace from `lldb`. It seems to work better than `gdb` in the Clang x64 environment.

```
(lldb) r
Process 948 launched: 'C:\msys64\home\nick\nvc\build\x-clang64\bin\nvc.exe' (x86_64)
Process 948 stopped
* thread #1, stop reason = Exception 0xc00000fd encountered at address 0x7fff729a1d7e
    frame #0: 0x00007fff729a1d7e _WORK.TEST.elab.dll`___chkstk_ms + 22
_WORK.TEST.elab.dll`___chkstk_ms:
->  0x7fff729a1d7e <+22>: testq  %rcx, (%rcx)
    0x7fff729a1d81 <+25>: subq   $0x1000, %rax             ; imm = 0x1000
    0x7fff729a1d87 <+31>: cmpq   $0x1000, %rax             ; imm = 0x1000
    0x7fff729a1d8d <+37>: ja     0x7fff729a1d77            ; <+15>
(lldb) bt
* thread #1, stop reason = Exception 0xc00000fd encountered at address 0x7fff729a1d7e
  * frame #0: 0x00007fff729a1d7e _WORK.TEST.elab.dll`___chkstk_ms + 22
    frame #1: 0x00007fff729a14ab _WORK.TEST.elab.dll`WORK.TEST.READ_MEM + 11
    frame #2: 0x00007fff729a1463 _WORK.TEST.elab.dll`WORK.TEST.P_PROC + 83
    frame #3: 0x00007ff65713f004 nvc.exe`jit_fastcall(j=0x000001598530c290, handle=<unavailable>, result=0x0000004809aff7d8, p1
=(integer = -1, pointer = 0xffffffffffffffff, real = NaN), p2=(integer = 1484034080768, pointer = 0x0000015987530000, real = 7.
3321025656506256E-312), tlab=0x0000015982980b48) at jit-core.c:657:7
    frame #4: 0x00007ff65713764a nvc.exe`async_run_process [inlined] run_process(m=0x00000159852f6e20, proc=0x0000015985308200)
 at model.c:1038:9
    frame #5: 0x00007ff657137559 nvc.exe`async_run_process(m=0x00000159852f6e20, arg=0x0000015985308200) at model.c:2594:4
    frame #6: 0x00007ff65712f3cc nvc.exe`model_cycle at model.c:375:7
    frame #7: 0x00007ff65712f3a5 nvc.exe`model_cycle(m=0x00000159852f6e20) at model.c:3095:4
    frame #8: 0x00007ff65712ed6c nvc.exe`model_run(m=0x00000159852f6e20, stop_time=9223372036854775807) at model.c:3142:7
    frame #9: 0x00007ff6570b455b nvc.exe`process_command [inlined] run_cmd(argc=2, argv=0x00000159829395b0, state=0x0000004809a
ffb40) at nvc.c:797:4
    frame #10: 0x00007ff6570b43d7 nvc.exe`process_command(argc=2, argv=0x00000159829395b0, state=0x0000004809affb40) at nvc.c:1
869:14
    frame #11: 0x00007ff6570b4136 nvc.exe`process_command(argc=<unavailable>, argv=0x0000015982939598, state=<unavailable>) at
nvc.c:0
    frame #12: 0x00007ff6570b17b5 nvc.exe`main(argc=<unavailable>, argv=<unavailable>) at nvc.c:2004:20
    frame #13: 0x00007ff6570b1315 nvc.exe`__tmainCRTStartup at crtexe.c:267:15
    frame #14: 0x00007ff6570b1366 nvc.exe`mainCRTStartup at crtexe.c:188:9
    frame #15: 0x00007fff7f087344 kernel32.dll`BaseThreadInitThunk + 20
    frame #16: 0x00007fff80de26b1 ntdll.dll`RtlUserThreadStart + 33
```

nickg, 2023-11-08:
This happens in Clang x64 only because the stack size there is 1MB whereas in the other MSYS2 environments it's 2MB. But the crash would still happen in MingW64 or in Linux if you increased the size of that array which NVC always allocates on the static when it has a constant size. So the fix here is just to limit the maximum size of stack allocations.

But there's still a problem where user code can cause a stack overflow that then crashes the whole program - e.g. if you called `read_mem` recursively. It would be better to handle that gracefully and report an error to the user with a VHDL backtrace but I'm not sure of the best way of doing that at the moment.

closed 2023-11-08 by nickg

avelure, 2023-11-08:
Ok, I see. a form of backtrace would be helpful as it was not so obvious to find the culprit in the large codebase.
I see the issue happens on GHDL as well, but not until the range is (0 to 2**29-1)

Does NVC support sparse memory allocation, in other words would it have helped if `(others => x"FF")` was removed to leave the memory uninitialized until assigned?

nickg, 2023-11-09:
> 
> Does NVC support sparse memory allocation, in other words would it have helped if `(others => x"FF")` was removed to leave the memory uninitialized until assigned?

No because even if you leave off the explicit initialiser VHDL still requires that every scalar sub-element be initialised to `type'left` so it would still have to write every location to `'U'` here.

commit 3a3e45e6 (2023-11-08): Limit maximum size of stack allocations to 64kB

Fixes #791