nickg/nvc #795: Basic PSL cover support

Blebowski, 2023-11-15:
Hi,

NVC currently does not support basic PSL cover statements, e.g. like so:

```
-- psl simul_a_and_b_cov : cover {signal_a = '1' and signal_b = '1'};
```

It would be handy to add support for at least basic PSL cover statements.
For the coverage engine, this would unlock possibility to add probably the last kind of coverage:
"Functional coverage".

Longer time ago, I looked into the PSL, but I didn't write anything in the "sem" or "lower" phases.
I only added parsing. I got stuck on the FSM conversion. I am sure how with an FSM to do, e.g.
an arbitrary sequence like so:

`{a;b} => {c;d}`

I did not want to do some "half" solution of generating an FSM that would need to be then reworked.
I know this type of sequence is not strictly simple subset, but  I think it would be handy to support
such sequences.

Anyway, if there could be at least basic support of `cover` with e.g. only a simple combinatorial expression,
this would unlock work on the functional coverage.

nickg, 2023-11-19:
I've added code generation for the `cover` directive and also connected it to the coverage database in ce6efd342. It should work for simple sequences as well as just boolean expressions.

Blebowski, 2023-11-19:
Thanks, I tried it and added support of excludes in pushed MR. It may have limited usability in case of functional coverage,
but I would still have it there.

closed 2023-11-19 by Blebowski