nickg/nvc #799: Incorrect assignment and access violation

avelure, 2023-11-20:
The below MWE gives an incorrect value, probably due to slicing of the return of to_unsigned.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

entity test1 is
end entity;

architecture rtl of test is
  constant C_IBC : natural := 500000; -- 0x07A120
  type t_slv_array is array (natural range <>) of std_logic_vector;
begin
  p_proc : process
    variable v_default_data  : t_slv_array(0 to 32)(15 downto 0);
  begin
    v_default_data(4) := std_logic_vector(to_unsigned(C_IBC/2, 32)(31 downto 16)) or "0000000001000000";
    assert v_default_data(4) = x"43" report to_string(v_default_data(4)) severity failure;
    wait;
  end process;
end architecture;
```
```
** Error: 0ms+0: Assertion Error: 0000000001100000
         C:\proj\public\minimal\test_vector_or.vhd:17
```

commenting out the literal and we can see that some of the bits are unassigned
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

entity test1 is
end entity;

architecture rtl of test is
  constant C_IBC : natural := 500000; -- 0x07A120
  type t_slv_array is array (natural range <>) of std_logic_vector;
begin
  p_proc : process
    variable v_default_data  : t_slv_array(0 to 32)(15 downto 0);
  begin
    v_default_data(4) := std_logic_vector(to_unsigned(C_IBC/2, 32)(31 downto 16));-- or "0000000001000000";
    assert v_default_data(4) = x"03" report to_string(v_default_data(4)) severity failure;
    wait;
  end process;
end architecture;
```
```
** Error: 0ms+0: Assertion Error: 001000011UUUUUUU
         C:\proj\public\minimal\test_vector_or.vhd:17
```

I also get an access violation
```
C:\proj\public\minimal$ lldb -- c://msys64//clang64//bin//nvc.exe --std=08 -e test -r test
(lldb) target create "c://msys64//clang64//bin//nvc.exe"
Current executable set to 'c:\msys64\clang64\bin\nvc.exe' (x86_64).
(lldb) settings set -- target.run-args  "--std=08" "-e" "test" "-r" "test"
(lldb) r
(lldb) Process 31600 launched: 'c:\msys64\clang64\bin\nvc.exe' (x86_64)
Process 31600 stopped
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ff7d2176192: Access violation reading location 0x00000001
    frame #0: 0x00007ff7d2176192 nvc.exe`jit_try_vcall [inlined] jit_thread_local at jit-core.c:135:8
   132     static __thread jit_thread_local_t *local = NULL;
   133  #endif
   134
-> 135     if (unlikely(local == NULL)) {
   136        local = xcalloc(sizeof(jit_thread_local_t));
   137        local->state = JIT_IDLE;
   138     }
(lldb) bt
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ff7d2176192: Access violation reading location 0x00000001
  * frame #0: 0x00007ff7d2176192 nvc.exe`jit_try_vcall [inlined] jit_thread_local at jit-core.c:135:8
    frame #1: 0x00007ff7d2176192 nvc.exe`jit_try_vcall [inlined] jit_transition(j=0x000001ea4e6b1340, from=JIT_RUNNING, to=0x1ea) at jit-core.c:615:33
    frame #2: 0x00007ff7d2176192 nvc.exe`jit_try_vcall(j=0x000001ea4e6b1340, f=0x0000000000000000, result=0x000001ea4e670000, args=0x0000000000000019, tlab=<unavailable>) at jit-core.c:661:7
```
But it goes away when I change the failure to a warning.

nickg, 2023-11-20:
> 
> I also get an access violation
> 

I think this has the same cause as #796 which is also fixed now.

closed 2023-11-20 by nickg

avelure, 2023-11-21:
> > I also get an access violation
> 
> I think this has the same cause as #796 which is also fixed now.

Yes, this seems to be fixed, thanks!

commit fdaaf982 (2023-11-20): Fix wrong result bounds when TO_UNSIGNED is constant folded

Fixes #799