nickg/nvc #804: Vital TimingChecksOn optimization

avelure, 2023-11-22:
Vital models normally have a boolean `TimingChecksOn` generic in their entity which is by default set to `true`. Depending on the models optimization it might not be used, but when it is used it is normally used to disable parts of the model that checks timing. 
Modelsim has a parameter which sets the generic to false. Would this be something that could be easily be implemented?

From Questa and Modelsim manual:
```+notimingchecks
(optional) Removes all timing check entries from the design as it is optimized. To disable
checks on individual instances, use the tcheck_set command.
Specifying vopt +notimingchecks or -GTimingChecks=<FALSE/TRUE> will fix the
TimingChecksOn generic value in VITAL models to FALSE for simulation. As a
consequence, using vsim +notimingchecks at simulation may not have any effect on the
simulation, depending on the optimization of the model.
```

```
(optional) Removes Verilog timing checks and disables VITAL timing checks for all
instances in the specified design (This option sets the generic TimingChecksOn to FALSE
for all VHDL Vital models with the Vital_level0 or Vital_level1 attribute. Does not affect
generics with the name TimingChecksOn on non-VITAL models.) By default, Verilog
timing check system tasks ($setup, $hold,…) in specify blocks are enabled. For VITAL, the
ASIC or FPGA vendor controls the timing check default, but most default to enabled.
```

Blebowski, 2023-11-22:
Hi @avelure , I would like to implement this as part of the SDF annotation support.

My work on the SDF annotation is stalled now, since I am very busy in my job duties and family stuff.

I hope that I could return to this work early next year, and do small step-by-step improvements.
Then, I will have time next year in summer, after we finish our current tape-out. I hope that I can
manage to finish the implementation next summer...

Blebowski, 2023-11-22:
If you are interested, feel free to extend the current MR. So far, it only implements parsing of
SDF files, annotation and configuration of timing checks is still TODO.

Blebowski, 2026-11-15:
This is now implemented in:
https://github.com/nickg/nvc/pull/1698

@avelure could you please give it a try with your VITAL models ?