nickg/nvc #806: Optimize bound checks

Anselmo95, 2023-11-26:
Hello,

In the code below nvc errors out during analysis while checking the length of **size_8** against **o**. This seams to happens only when the boolean condition involves a constant, if for example we remove the package and move **W** in the generics of **opt** the code works fine. 

Would it be possible to avoid performing the bound checks when the boolean condition is always constant ?

```vhdl
package a is
    constant W : natural := 9;
end package a;

library ieee;
use ieee.std_logic_1164.all;

use work.a.all;

entity opt is
    port (
        o : out std_logic_vector(W-1 downto 0)
    );
end entity opt;

architecture rtl of opt is
    signal size_8 : std_logic_vector(7 downto 0);
    signal size_9 : std_logic_vector(8 downto 0);
begin
    
    o <= size_9 when W = 9 else size_8; 
    
end architecture rtl;
```

nickg, 2023-11-26:
This is a side-effect of having to keep the signal assignments around to create drivers. Anyway should be handled better now.

Anselmo95, 2023-11-27:
Thanks a lot for the explanation and the quick fix !

commit ff41e268 (2023-11-26): Avoid bounds check failure on never-taken branch

Fixes #806