nickg/nvc #813: Githash in version output

avelure, 2023-12-12:
It looks like that after the change to v1.12-devel the information of the githash that the version was built from disappeared, is this intentional?

```
nvc 1.12-devel () (Using LLVM 15.0.7)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```
previously
```
nvc 1.11-devel (1.10.0.r331.g3f21c263) (Using LLVM 15.0.7)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

nickg, 2023-12-13:
It's not intentional. The command the makefile runs to generate that is `git describe --long | sed 's/^r\(.*\)/\1/; s/\([^-]*-g\)/r\1/;s/-/./g'` could you try running that and/or `git describe --long` and see what it prints?

avelure, 2023-12-14:
I think this is related to a change in my docker build script.
I changed the repo checkout to be `git clone --depth=1 https://github.com/nickg/nvc.git` to speed up the checkout.
As this doesn't check out the commits since the last tag, then I guess git can not count them to present them with `describe`
```
root@4d64c00ba9d9:/tmp/nvc# git describe --long
fatal: No names found, cannot describe anything.
```

nickg, 2023-12-16:
Found a simple solution is to just add `--always` to the `git describe` command. Now it prints:

```
nvc 1.12-devel (2025713e16a9) (Using LLVM 14.0.6)
```

commit 92d426ae (2023-12-16): Include Git SHA in version output when no tags available

Fixes #813
commit cb61553f (2023-12-16): Include Git SHA in version output when no tags available

Fixes #813
commit f439f4cd (2023-12-16): Include Git SHA in version output when no tags available

Fixes #813
closed 2023-12-16 by nickg