nickg/nvc #816: VHDL entity with a name using extended idenfier

jorkrohn, 2023-12-19:
It seems like analysing an entity with a name using extended_identifier doesn't work. At least in windows using MSYS2
I encountered this problem while trying to compile a netlist for an IP generated by Vivado. I made a simple example below.

Trying to analyse the entity:
```
library ieee;
use ieee.std_logic_1164.all;

entity \test\ is
end entity \test\;
```

gives the following error message:
`** Fatal: failed to create WORK.\test\ in library WORK`

I'm guessing it's because nvc tries to create a file with this name, which is not a legal name in windows.

nickg, 2023-12-20:
Yes those file names probably need special handling/escaping on Windows.

nickg, 2023-12-20:
There's a similar problem on macOS which even though it allows `/` in file names is case-insensitive so `/foo/` and `/Foo/` would collide. Anyway should be fixed now. Thanks for reporting!

commit c739b978 (2023-12-20): Avoid illegal characters in design unit file names. Fixes #816
commit 34b03756 (2023-12-22): Avoid illegal characters in design unit file names. Fixes #816
closed 2023-12-20 by nickg