nickg/nvc #820: VHDL 93: Unexpected X when concatenating data from generate blocks.

aqeelmahroof, 2024-01-02:
Hi Nick,

I'm currently running a simulation for the following code, where data is concatenated, and the limits are specified through a function.
However, during the simulation, I've noticed that the simulator is introducing unexpected 'X' values.
```vhdl
ibrary ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity check_rtl is
generic (
  G_NUM_ARR                     : integer := 4;
  G_DWIDTH                      : integer := 4*8
);
port (
  clk                          : in std_logic;
  o_dout                       : out std_logic_vector(G_DWIDTH-1 downto 0) := (others => '0')
);
end check_rtl;

architecture rtl of check_rtl is
  type t_ram_property is array (natural range<>) of natural;

  function num_arr (g_num_arr : natural) return t_ram_property is
    variable ram_dwidth : t_ram_property(0 to g_num_arr-1) := (others => 8);
  begin
    return(ram_dwidth);
  end function;

  function f_bit_low (ram_dwidth : t_ram_property) return t_ram_property is
    variable ram_bit_low : t_ram_property(ram_dwidth'range) := (others => 0);
  begin
    for i in ram_dwidth'range loop
      if(i=0) then
        ram_bit_low(i) := 0;
      else
        ram_bit_low(i) := ram_bit_low(i-1) + ram_dwidth(i-1);
      end if;
    end loop;
    return(ram_bit_low);
  end function;
  
  function f_bit_high (ram_dwidth : t_ram_property) return t_ram_property is
    variable ram_bit_high : t_ram_property(ram_dwidth'range) := (others => 0);
  begin
    for i in ram_dwidth'range loop
      if(i=0) then
        ram_bit_high(i) := -1 + ram_dwidth(i);
      else
        ram_bit_high(i) := ram_bit_high(i-1) + ram_dwidth(i);
      end if;
    end loop;
    --
    return(ram_bit_high);
  end function;  
  
  constant C_RAM_DWIDTH        : t_ram_property := num_arr(G_NUM_ARR);
  constant C_RAM_BIT_LOW       : t_ram_property := f_bit_low(C_RAM_DWIDTH);
  constant C_RAM_BIT_HIGH      : t_ram_property := f_bit_high(C_RAM_DWIDTH);
  
  signal s_clk : std_logic := '0';

begin

  p_clk: process
  begin
    wait for 100 ns;
    s_clk <= '1';
    wait for 100 ns;
    s_clk <= '0';
    wait for 100 ns;
    s_clk <= '1';
    wait for 100 ns;
    s_clk <= '0';
  end process;

  gen_arr: for i in 0 to G_NUM_ARR-1 generate
    signal  s_ram_reg_datab   : unsigned(C_RAM_DWIDTH(i)-1 downto 0) := (others => '0');
  begin

    p_dummy: process(s_clk)
      begin
       if (rising_edge(s_clk)) then
        s_ram_reg_datab <= s_ram_reg_datab + 1;
      end if;
    end process;
    --
    gen_reg: if(True) generate
      o_dout(C_RAM_BIT_HIGH(i)  downto C_RAM_BIT_LOW(i)) <= std_logic_vector(s_ram_reg_datab);
    end generate;
  end generate gen_arr;

end architecture rtl;
``` 

Im expecting thease values 
[image]
But its resulting in X values 
[image]

Steps to run:

1. `nvc --std=93 -a check_rtl.vhd`
2. `nvc   -e  check_rtl --no-save -O0 -r  --format=fst --wave=dump.fst --stop-time=1us`

Version 
nvc --version
nvc 1.12-devel (1.11.0.r19.gc739b978) (Using LLVM 14.0.0)
(platform Ubuntu 22.04.3 LTS) 

Best Regards,
Aqeel Mahroof.

nickg, 2024-01-02:
Should be fixed now, thanks for reporting.

closed 2024-01-02 by nickg

aqeelmahroof, 2024-01-04:
Thanks Nick, That was pretty quick! 👍 . 
Ran the sims with the fix, its working fine.

commit 630e17d0 (2024-01-02): Fix longest static prefix calculation with constant array reference

Fixes #820
commit 8043f60f (2024-01-03): Fix longest static prefix calculation with constant array reference

Fixes #820