nickg/nvc #821: Fatal: Write to object in frozen arena

Anselmo95, 2024-01-02:
Hello nickg, 

While elaborating a design nvc errors out with the following trace:
```
** Fatal: Write to object in frozen arena WORK.ENT-RTL [address=0x7f34d4c05398]
etc...
[0x4d1bd3] ./../src/diag.c:1015 diag_femit
[0x4d1d91] ./../src/diag.c:1039 diag_emit
[0x412dbc] ./../src/util.c:585 fatal_trace
[0x483f43] ./../src/object.c:358 check_frozen_object_fault
[0x4135fb] ./../src/util.c:861 signal_handler
[0x7f38ecd19cef] (/usr/lib64/libpthread-2.28.so) 
[0x483845] ./../src/object.c:239 obj_array_add
[0x447140] ./../src/tree.c:509 tree_array_add.lto_priv.62
[0x44965b] ./../src/tree.c:883 tree_add_stmt
[0x46de58] ./../src/simp.c:747 simp_make_dummy_drivers
[0x46e7ee] ./../src/simp.c:946 simp_if
[0x4700e5] ./../src/simp.c:1552 simp_tree
[0x484d70] ./../src/object.c:675 object_rewrite_iter
[0x48545a] ./../src/object.c:780 object_rewrite
[0x4852b2] ./../src/object.c:749 object_rewrite
[0x4852b2] ./../src/object.c:749 object_rewrite
[0x44bd69] ./../src/tree.c:1286 tree_rewrite
[0x47058f] ./../src/simp.c:1695 simplify_global
[0x4698fe] ./../src/elab.c:1326 elab_instance
[0x46a992] ./../src/elab.c:1651 elab_stmts
[0x46a2af] ./../src/elab.c:1524 elab_for_generate
[0x46a9c2] ./../src/elab.c:1657 elab_stmts
[0x46a5d5] ./../src/elab.c:1578 elab_if_generate
[0x46a9d7] ./../src/elab.c:1660 elab_stmts
[0x469982] ./../src/elab.c:1337 elab_instance
[0x46a992] ./../src/elab.c:1651 elab_stmts
[0x46b2ea] ./../src/elab.c:1823 elab_top_level
[0x46b5a1] ./../src/elab.c:1883 elab
[0x40bb0f] ./../src/nvc.c:456 elaborate
[0x40ee0f] ./../src/nvc.c:1905 process_command
[0x40f233] ./../src/nvc.c:2042 main
```

This seams a regression caused by this commit in particular https://github.com/nickg/nvc/commit/ff41e26881d1a468c707f43ec808fae03ced6656.

At the moment I don't have a small code snippet to reproduce the issue and cannot share the original code.
The entity in the backtrace is inside many nested `if generate` statements and that's the only thing, I guess, could be releated to this issue.

I will keep trying to create a reproducer but do you have some suggestion on how I could proceed?

nickg, 2024-01-02:
Thanks for providing the back trace, I think I've managed to work it out from that. Here's a minimal reproducer:

```vhdl
entity issue821 is
end entity;

architecture test of issue821 is
    signal x : integer;
begin

    g: for i in 1 to 2 generate
    begin
        p: process is
        begin
            if i > 1 then
                x <= 1;
            elsif x < 100 then
                x <= 2;
            end if;
            wait;
        end process;
    end generate;

end architecture;
```

nickg, 2024-01-02:
Could you try again with the latest master branch? I think it should be fixed now.

sean-anderson-seco, 2024-01-02:
I was just about to report this bug as well. I can confirm that the latest master fixes this. Thanks.

Anselmo95, 2024-01-02:
I confirm too, the problem is fixed now. Thanks a lot!

closed 2024-01-02 by Anselmo95