nickg/nvc #824: Record Resolution and Upgrade to version 1.11.2

JimLewis, 2024-01-07:
I am now using mingw-w64-x86_64-nvc-1.11.2.git20240104.32ebd286-1-any.pkg.tar.zst

I upgraded to resolve previously reported issues, however, this release introduces a new issue.

I have a record:
```
  type CpuRecType is record
    Clk         : std_logic ;
    nAds        : std_logic ;
    Addr        : CpuAddrType ; 
    Data        : CpuDataType ; 
    Wr_nRd      : std_logic ;
    nRdy        : std_logic ;
  end record ;
```
The testbench initializes the record by doing:
`  signal CpuRec              : CpuRecType := INIT_SUB_CPU_REC ;`

Then I have a subprogram that has CpuRec as an inout.   
```
  procedure CpuWrite (
    signal   CpuRec      : InOut CpuRecType ;
             iAddr       : In    std_logic_vector ;
             iData       : In    std_logic_vector 
  ) ;
```

The nRdy element is not driven by the CpuWrite is now a U.
If in the process that calls the subprogram I explicitly assign nRdy to a Z,
`    CpuRec.nRdy <= 'Z' ;`

then everything works again.  

So apparently the initialization of the record is not effective for the process for some reason.   

It worked in past versions of NVC.   It works in all other simulators.

This is from our class materials.   I can provide a running example under NDA if needed.

nickg, 2024-01-07:
What's the value of `INIT_SUB_CPU_REC`?

JimLewis, 2024-01-07:
```
 Constant INIT_SUB_CPU_REC : CpuRecType := (
    Clk      => 'Z', 
    nAds     => 'Z', 
    Addr     => (others => 'Z'), 
    Data     => (others => 'Z'),
    Wr_nRd   => 'Z', 
    nRdy     => 'Z'
  ) ;
```

JimLewis, 2024-01-07:
Also
```
 subtype  CpuAddrType is std_logic_vector(20 downto 0) ; 
subtype  CpuDataType is std_logic_vector(15 downto 0) ; 
```

nickg, 2024-01-07:
Does the `CpuRec` signal have any drivers besides that process?

JimLewis, 2024-01-07:
Yes.   And they each have proper resolution functions.

nickg, 2024-01-08:
Is it also connected to an `out` port? Here's a minimal example that gets the wrong result:

```vhdl
entity test is
end entity;

library ieee;
use ieee.std_logic_1164.all;

architecture test of test is
    type t_rec is record
        x, y : std_logic;
    end record;

    procedure update (signal r : inout t_rec) is
    begin
        r.y <= '1';
    end procedure;

    signal s : t_rec := ('Z', 'Z');
begin

    p1: process is
    begin
        --s.x <= 'Z';
        wait for 1 ns;
        update(s);
        wait for 1 ns;
        assert s.x = 'Z';
        wait;
    end process;

    b1: block is
        port ( o : out t_rec );
        port map ( s );
    begin
        o <= ('Z', 'Z');
    end block;

end architecture;
```

JimLewis, 2024-01-08:
Almost.   On the DUT (your block statement), nRdy is an output as a regular signal and then the port map connects to the record. 

Hence the port map of the DUT includes 
```
port map(
   nRdy => CpuRec.nRdy, 
   . . .   
```

I suspect they are similar enough that this will work as a minimal example.

nickg, 2024-01-08:
@JimLewis can you try the package from here and see if it resolves the issue? https://github.com/nickg/nvc/actions/runs/7453772337/artifacts/1155340913

JimLewis, 2024-01-08:
That got it.   The test cases that were failing are now passing.  Now for the bigger regression.   I am thinking that unless this broke something, all OSVVM tests should now pass.   Let you know in a couple of hours

JimLewis, 2024-01-09:
Build: sim_RunAllTests PASSED,  Passed: 783,  Failed: 0,  Skipped: 4,  Analyze Errors: 0,  Simulate
Errors: 0

The tests that are skipped have std.env.stop and need a script that can do `run -all ; run -all ; run -all ` so that when they hit stop, then can continue running more.   It is for testing our capability to stop a simulation when a stop count is hit.

JimLewis, 2024-01-09:
Hopefully the new code will be an official part of the next release?

closed 2024-01-09 by JimLewis

nickg, 2024-01-09:
> Hopefully the new code will be an official part of the next release?

I'll include it in a 1.11.3 maintenance release at some point