nickg/nvc #826: aggregate and array directions when the aggregate is in an expression

JimLewis, 2024-01-10:
So I put together some examples of where I hate aggregates.   A few failed in NVC.   The test results are below.   It looks like NVC had issues when the aggregate in the expression tried to hint that is is structured as downto direction where the current VHDL rules say that the direction in an expression follows the direction of the array index definition - which is typically natural which has a direction 0 to ...

Test Result is:
%% Log    ALWAYS    in Default,   Aggregates in assignments follow direction of array with 7 downto 0 at 0 ns
%% Log    PASSED    in TbID1,   SLV := (others => '0') Received : 00 at 0 ns
%% Log    PASSED    in Default,   SLV := (7 downto 0 => '0') Received : 00 at 0 ns
%% Log    PASSED    in Default,   SLV := ('0','0','0','0','0','0','0','0') Received : 00 at 0 ns
%% Log    PASSED    in Default,   SLV := ((7|6 =>'0', 5 downto 0 =>'0')) Received : 00 at 0 ns
%% Log    PASSED    in Default,   SLV := (7|6 =>'1', 5 downto 0 =>'0') Received : C0 at 0 ns
%% Log    PASSED    in Default,   SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') Received : C0 at 0 ns
%% Log    ALWAYS    in Default,   Aggregates in assignments follow direction of array with 0 to 7 at 0 ns
%% Log    PASSED    in Default,   SLVR := (7|6 =>'1', 5 downto 0 =>'0') - value reverses when array direction does Received : 03 at 0 ns
%% Log    PASSED    in Default,   SLVR := (0 to 5 =>'0', 6 =>'1',  7 =>'1') - value reverses when array direction does Received : 03 at 0 ns
%% Log    ALWAYS    in Default,   Aggregates in expressions use direction 0 to 7 using not at 0 ns
%% Log    PASSED    in Default,   SLV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1')) Received : FC at 0 ns
%% Log    PASSED    in Default,   UV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1')) Received : FC at 0 ns
%% Alert  ERROR     in Default,   SLV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) - Direction Reversed Received : 3F ?= Expected : FC at 0 ns
%% Alert  ERROR     in Default,   UV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) - Direction Reversed Received : 3F ?= Expected : FC at 0 ns
%% Log    ALWAYS    in Default,   Aggregates in expressions use direction 0 to 7 using or at 0 ns
%% Alert  ERROR     in Default,   SLV := ZERO or (7|6 =>'1', 5 downto 0 =>'0') Received : C0 ?= Expected : 03 at 0 ns
%% Log    PASSED    in Default,   SLV := ZERO or (0 to 5 =>'0', 6 =>'1',  7 =>'1') Received : 03 at 0 ns
%% Log    PASSED    in Default,   SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO Received : 03 at 0 ns
%% Alert  ERROR     in Default,   UV := ZERO_UV or (7|6 =>'1', 5 downto 0 =>'0') Received : C0 ?= Expected : 03 at 0 ns
%% Log    PASSED    in Default,   UV := ZERO_UV or (0 to 5 =>'0', 6 =>'1',  7 =>'1') Received : 03 at 0 ns
%% Log    PASSED    in Default,   UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO_UV  Received : 03 at 0 ns
%% Alert  ERROR     in Default,   UV := ZERO_UV + (7|6 =>'1', 5 downto 0 =>'0') Received : C0 ?= Expected : 03 at 0 ns
%% Log    PASSED    in Default,   UV := ZERO_UV + (0 to 5 =>'0', 6 =>'1',  7 =>'1') Received : 03 at 0 ns
%% Log    PASSED    in Default,   UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') + ZERO_UV  Received : 03 at 0 ns
%% Log    ALWAYS    in Default,   numeric_std and std_logic_1164 functions return different array directions at 0 ns
%% Log    PASSED    in Default,   SL := "not"(not(std_logic_vector'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7) Received : 1 at 0 ns
%% Log    PASSED    in Default,   SL := "not"(not(unsigned'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7) - Array returned as 7 downto 0, leaving 1 on RHS Received : 0 at 0 ns

The test case is (has OSVVM stuff)
```
library IEEE ;
use ieee.std_logic_1164.all ; 
use ieee.numeric_std.all ; 
-- use ieee.numeric_std_unsigned.all ; 

library OSVVM ; 
context OSVVM.OsvvmContext ; 

entity aggregate is 
end entity aggregate ; 
architecture test of aggregate is 
  constant ZERO : std_logic_vector(7 downto 0) := X"00" ;
  constant ZERO_UV : unsigned(0 to 7) := X"00" ;
  signal TestDone : BarrierType ; 
begin
  ControlProc : process
  begin
    SetTestName("aggregate") ; 
    SetLogEnable(PASSED, TRUE) ; 
    WaitForBarrier(TestDone, 1 ms) ;
    EndofTestReports ;  -- nothing to report here
    std.env.stop ; 
    wait ;
  end process ControlProc ; 
 
 
  TestProc : process
    variable SLV : std_logic_vector(7 downto 0) ;
    variable SLVR : std_logic_vector(0 to 7) ;
    variable UV : unsigned(0 to 7) ;
    variable SL : std_logic ; 
    variable TbId : AlertLogIDType ; 

  begin
    wait for 0 ns ;  -- Allow SetLogEnable to be set 
    wait for 0 ns ;
    TbID := NewID("TbID1") ;
    
Log("Aggregates in assignments follow direction of array with 7 downto 0") ; 
    SLV := (others => '0') ; 
    AffirmIfEqual(TbID, SLV, X"00", "SLV := (others => '0')") ;

    SLV := (7 downto 0 => '0') ; 
    AffirmIfEqual(SLV, X"00", "SLV := (7 downto 0 => '0')") ;

    SLV := ('0','0','0','0','0','0','0','0') ; 
    AffirmIfEqual(SLV, X"00", "SLV := ('0','0','0','0','0','0','0','0')") ;

    SLV := ((7|6 =>'0', 5 downto 0 =>'0')) ; 
    AffirmIfEqual(SLV, X"00", "SLV := ((7|6 =>'0', 5 downto 0 =>'0'))") ;

    SLV := (7|6 =>'1', 5 downto 0 =>'0') ; 
    AffirmIfEqual(SLV, X"C0", "SLV := (7|6 =>'1', 5 downto 0 =>'0')") ;

    SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') ; 
    AffirmIfEqual(SLV, X"C0", "SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1')") ;

Log("Aggregates in assignments follow direction of array with 0 to 7") ; 
    SLVR := (7|6 =>'1', 5 downto 0 =>'0') ; 
    AffirmIfEqual(SLVR, X"03", "SLVR := (7|6 =>'1', 5 downto 0 =>'0') - value reverses when array direction does") ;

    SLVR := (0 to 5 =>'0', 6 =>'1',  7 =>'1') ; 
    AffirmIfEqual(SLVR, X"03", "SLVR := (0 to 5 =>'0', 6 =>'1',  7 =>'1') - value reverses when array direction does") ;

Log("Aggregates in expressions use direction 0 to 7 using not") ; 
    SLV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1')) ; 
    AffirmIfEqual(SLV, X"FC", "SLV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1'))") ;

    UV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1')) ; 
    AffirmIfEqual(UV, X"FC", "UV := not((0 to 5 =>'0', 6 =>'1',  7 =>'1'))") ;

    SLV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) ; 
    AffirmIfEqual(SLV, X"FC", "SLV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) - Direction Reversed") ;

    UV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) ; 
    AffirmIfEqual(UV, X"FC", "UV := not((7 =>'1', 6 =>'1',  5 downto 0 =>'0')) - Direction Reversed") ;

Log("Aggregates in expressions use direction 0 to 7 using or") ; 
    SLV := ZERO or (7|6 =>'1', 5 downto 0 =>'0') ; 
    AffirmIfEqual(SLV, X"03", "SLV := ZERO or (7|6 =>'1', 5 downto 0 =>'0')") ;

    SLV := ZERO or (0 to 5 =>'0', 6 =>'1',  7 =>'1') ; 
    AffirmIfEqual(SLV, X"03", "SLV := ZERO or (0 to 5 =>'0', 6 =>'1',  7 =>'1')") ;

    SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO ; 
    AffirmIfEqual(SLV, X"03", "SLV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO") ;

    UV := ZERO_UV or (7|6 =>'1', 5 downto 0 =>'0') ; 
    AffirmIfEqual(UV, X"03", "UV := ZERO_UV or (7|6 =>'1', 5 downto 0 =>'0')") ;

    UV := ZERO_UV or (0 to 5 =>'0', 6 =>'1',  7 =>'1') ; 
    AffirmIfEqual(UV, X"03", "UV := ZERO_UV or (0 to 5 =>'0', 6 =>'1',  7 =>'1')") ;

    UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO_UV ; 
    AffirmIfEqual(UV, X"03", "UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') or ZERO_UV ") ;

    UV := ZERO_UV + (7|6 =>'1', 5 downto 0 =>'0') ; 
    AffirmIfEqual(UV, X"03", "UV := ZERO_UV + (7|6 =>'1', 5 downto 0 =>'0')") ;

    UV := ZERO_UV + (0 to 5 =>'0', 6 =>'1',  7 =>'1') ; 
    AffirmIfEqual(UV, X"03", "UV := ZERO_UV + (0 to 5 =>'0', 6 =>'1',  7 =>'1')") ;

    UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') + ZERO_UV ; 
    AffirmIfEqual(UV, X"03", "UV := (0 to 5 =>'0', 6 =>'1',  7 =>'1') + ZERO_UV ") ;
    
Log("numeric_std and std_logic_1164 functions return different array directions") ; 
    SL := "not"(not(std_logic_vector'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7) ; 
    AffirmIfEqual(SL, '1', "SL := ""not""(not(std_logic_vector'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7)") ;

    SL := "not"(not(unsigned'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7) ; 
    AffirmIfEqual(SL, '0', "SL := ""not""(not(unsigned'(0 to 5 =>'0', 6 =>'1',  7 =>'1')))(7) - Array returned as 7 downto 0, leaving 1 on RHS") ;

    WaitForBarrier(TestDone) ;
    wait ; 

  end process TestProc ; 
end architecture test ; 
```

nickg, 2024-01-12:
This is a mistake in the implementation of the 2008 changes (it works correctly with `--std=1993).

> If the aggregate does not appear in one of the contexts in the preceding list and an element association in the aggregate has a choice that is a discrete range and an expression that is of the type of the aggregate, then the direction of the index range of the aggregate is that of the discrete range.

The "... and an expression that is of the type of the aggregate" check is missing.

JimLewis, 2024-01-12:
Are you saying the direction changed between 1993 and 2008 or just the rules changed so much that you missed one of the updates?

nickg, 2024-01-12:
The paragraph in 9.3.3.3 that starts

> The direction of the index range of an array that does not have an others choice are determined as follows:

was not present in earlier standards. But the implementation is NVC is wrong: the direction in this case is the same in 1993 and 2008.

closed 2024-01-12 by nickg

JimLewis, 2024-01-12:
Thanks.   If only I could get this sort of response out of commercial vendors.   Instead, a week later I get an email - our tool does not crash so I don't understand why you are calling it an error.

nickg, 2024-01-12:
Updated package here: https://github.com/nickg/nvc/actions/runs/7507192815/artifacts/1166318490

commit 3d08d695 (2024-01-12): Fix incorrect aggregate direction with range association. Fixes #826
commit 7d29b50a (2024-01-12): Fix incorrect aggregate direction with range association. Fixes #826