nickg/nvc #827: Aggregates and VHDL-2008 stuff

JimLewis, 2024-01-10:
Maybe related to the previous one, but probably not so I filed it separately.   The following test case produces the following.   Something is very wrong with SLV
nvc --std=08 -H 128m --work=DefaultLib:C:/tools/VHDL_LIBS/nvc-1.11.2/DEFAULTLIB.08 -L C:/tools/VHDL_LIBS/nvc-1.11.2 -r --ieee-warnings=off aggregate_slice_2008
%% Log    ALWAYS    aggregate_slice_2008s in assignments follow direction of array at 0 ns
%% Alert  ERROR     SLV := (3 downto 0 => X"C", others => '0') Received : 01 ?= Expected : 0C at 0 ns
%% Log    PASSED    SLV := (7 downto 4 => '0') & X"C" Received : 0C at 0 ns
%% DONE   FAILED   aggregate_slice_2008  Total Error(s) = 1  Failures: 0  Errors: 1  Warnings: 0  Passed: 1  Affirmations Checked: 2  at 0 ns

Seems like the first assignment made it unhappy.
```
library IEEE ;
use ieee.std_logic_1164.all ; 
use ieee.numeric_std.all ; 
-- use ieee.numeric_std_unsigned.all ; 

library OSVVM ; 
context OSVVM.OsvvmContext ; 

entity aggregate_slice_2008 is 
end entity aggregate_slice_2008 ; 
architecture test of aggregate_slice_2008 is 

  signal TestDone : BarrierType ; 
begin
  ControlProc : process
  begin
    SetTestName("aggregate_slice_2008") ; 
    SetLogEnable(PASSED, TRUE) ; 
    WaitForBarrier(TestDone, 1 ms) ;
    EndofTestReports ;  -- nothing to report here
    std.env.stop ; 
    wait ;
  end process ControlProc ; 
 
 
  TestProc : process
    variable SLV : std_logic_vector(7 downto 0) ;
    variable UV : unsigned(0 to 7) ;
    variable SL : std_logic ; 

  begin
    wait for 0 ns ;  -- Allow SetLogEnable to be set 
    wait for 0 ns ;
    
Log("aggregate_slice_2008s in assignments follow direction of array") ; 
    SLV := (3 downto 0 => X"C", others => '0') ; 
    AffirmIfEqual(SLV, X"0C", "SLV := (3 downto 0 => X""C"", others => '0')") ;

-- Does not compile - Yeah!
--    SLV := (others => X"C") ; 
--    AffirmIfEqual(SLV, X"CC", "SLV := (3 downto 0 => X""C"", others => '0')") ;

    SLV := (7 downto 4 => '0') & X"C" ; 
    AffirmIfEqual(SLV, X"0C", "SLV := (7 downto 4 => '0') & X""C""") ;

    WaitForBarrier(TestDone) ;
    wait ; 

  end process TestProc ; 
end architecture test ; 
```

nickg, 2024-01-12:
I think this has the same cause as #826, the test passes now anyway.

JimLewis, 2024-01-12:
I installed updates for #826 and this one still failed.   The other ones PASSED so I am fairly confident that I actually did the update :)

nickg, 2024-01-12:
OK looks like I ran the other test case again by mistake.

nickg, 2024-01-13:
https://github.com/nickg/nvc/actions/runs/7512964235/artifacts/1167294753

closed 2024-01-13 by nickg

JimLewis, 2024-01-13:
I concur:
Build: SandBox_Aggregates PASSED,  Passed: 4,  Failed: 0,  Skipped: 0,  Analyze Errors: 0,  Simulate Errors: 0