nickg/nvc #830: invalid type kind T_ARRAY in range_of a bunch of addresses

Galactic-Explorer, 2024-01-15:
I am getting a fatal error but unable to understand what caused it or where, and thus I cannot create a reproducible minimal version.

On MSYS2 the error says:

`** Fatal: invalid type kind T_ARRAY in range_of
[00007ff6f1e643d0]
[00007ff6f1f114fb]
[00007ff6f1e63183]
[00007ff6f1eca05c]
[00007ff6f1edf733]
[00007ff6f1eeb0c0]
[00007ff6f1effd19]
[00007ff6f1efff73]
[00007ff6f1f0cf09]
[00007ff6f1f099f0]
[00007ff6f1eba1c6]
[00007ff6f1ebe343]
[00007ff6f1eb619c]
[00007ff6f1e5924d]
[00007ff6f1e52cf1]
[00007ff6f1e41315]
[00007ff6f1e41366]
[00007fffc1c07344] BaseThreadInitThunk+0x14
[00007fffc2a626b1] RtlUserThreadStart+0x21
`

While on WSL2 the error says: 

** Fatal: invalid type kind T_ARRAY in range_of
[0x55a1b0290fe3]
[0x55a1b01cd4a2]
[0x55a1b0238697]
[0x55a1b024aaea]
[0x55a1b0253bc7]
[0x55a1b0261fdb]
[0x55a1b02627ae]
[0x55a1b02700fd]
[0x55a1b02296a2]
[0x55a1b022d335]
[0x55a1b0230a3c]
[0x55a1b02328b7]
[0x55a1b0234d6e]
[0x55a1b01d6caf]
[0x55a1b01d424f]
[0x55a1b01cae00]
[0x7f4d18973d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x7f4d18973e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) __libc_start_main
[0x55a1b01cc1e4]

nickg, 2024-01-15:
Can you try running it under GDB to get a backtrace? E.g. `gdb --args nvc [..options..]` and then enter `bt` when it crashes.

nickg, 2024-01-26:
Please also try with the latest master branch as the error fixed in #842 was similar to this.

closed 2024-03-10 by nickg