nickg/nvc #833: Working directory in search paths for libraries

m-kru, 2024-01-16:
Shouldn't the working directory by default be included in the search paths for libraries? During the analysis, for `--work` the `path` is optional, and by the default it is a directory in the current working directory. However, during the elaboration one must provide `-L .` for libraries in the current working directory to be found.

nickg, 2024-01-16:
It's the same for analysis: you have to pass `-L .` for any library that is not the working library.

m-kru, 2024-01-16:
I think we are not on the same page here. Let's suppose I have analyzed some entity into library `lib`. The `lib` directory was created in the working directory. However, I am not able to elaborate the top entity without adding `-L .`, even though the `lib` directory is in the working directory. In other words, the current working directory is not included in the libraries search paths during elaboration and run. However, when I analyze a file and I provide only the library name (without library path) the library directory is created in the working directory.

nickg, 2024-01-16:
That's because the working library is implicitly a directory called `work` if you don't specify it with `--work=`. So in your example you need to do `nvc --work=lib -e toplev`.

m-kru, 2024-01-16:
`$PREFIX`? Is this nvc specific environment variable?

It seemed reasonable to me to search for libraries in the current working directory by default. However, if it is designed in a different way, then ok.

nickg, 2024-01-16:
> `$PREFIX`? Is this nvc specific environment variable?

I just meant the installation prefix (e.g. `/usr/local` or whatever you passed as `--prefix=` when you ran `configure`).

> 
> It seemed reasonable to me to search for libraries in the current working directory by default. However, if it is designed in a different way, then ok.

The problem with searching the current directory by default is that you may get different behaviour depending on which directory you run the `nvc` command in. If you specify the `--work=` argument as an absolute path, and pass absolute paths to source files, then there wouldn't be any dependency on the working directory and the build will be completely reproducible. Adding `-L .` by default would lose that and there's no way to remove a library from the search list.

If a project is big enough that it's split over multiple libraries then it will be built with a script or makefile rather than typing the command by hand so passing `-L .` explicitly is not a big burden and as a general rule I prefer behaviours like this to be explicit rather implicit.