nickg/nvc #835: Slicing type conversion

avelure, 2024-01-16:
The code below gives no analysis error, but instead gives an elaboration error.
GHDL gives `type conversion cannot be indexed or sliced`
Modelsim gives `Prefix of slice name cannot be type conversion (UNSIGNED) expression.`

On a side note, is there a way I can do the conversion in the port map without creating an intermediary unsigned signal? 
For reasons I do not understand the output of the unsigned conversion is not constrained, even though the input is, so removing the slicing gives another error.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_port is
  port(
    sig : in unsigned
  );
end entity test_port;
architecture beh of test_port is
begin
end architecture beh;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;
architecture beh of test is
  signal sig_int : std_logic_vector(4 downto 0);
begin
  i_ent : entity work.test_port
  port map (
    sig => unsigned(sig_int)(sig_int'range)
  );
end architecture;
```

```
** Fatal: tree kind T_WAVEFORM does not have item I_TYPE
    > c:\proj\public\minimal\test_port_map_unsigned.vhd:25
    |
 25 |     sig => unsigned(sig_int)(sig_int'range)
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[00007FF6964C27F0]
[00007FF6965A4527] vhpi_is_printable+0xc97
[00007FF69656503A]
[00007FF696516E10]
[00007FF69651B135]
[00007FF69651316F]
[00007FF6964B91E1]
[00007FF6964B2CA1]
[00007FF6964A1315]
[00007FF6964A1366]
[00007FF9A3C5257D] BaseThreadInitThunk+0x1d
[00007FF9A444AA58] RtlUserThreadStart+0x28
```

nickg, 2024-01-16:
> The code below gives no analysis error, but instead gives an elaboration error. GHDL gives `type conversion cannot be indexed or sliced` Modelsim gives `Prefix of slice name cannot be type conversion (UNSIGNED) expression.`

I've probably missed some rule about this but I can't immediately find it in the LRM.

> 
> For reasons I do not understand the output of the unsigned conversion is not constrained, even though the input is, so removing the slicing gives another error.

This behaviour is required by the LRM. From 5.3.2.2 in the 2008 version:

> For an interface object or subelement whose mode is in, inout or linkage, **if the actual part includes a conversion function or a type conversion, then the result type of that function or the type mark of the type conversion shall define a constraint for the index range corresponding to the index range of the object, and the index range of the object is obtained from that constraint**; otherwise, the index range is obtained from the object or value denoted by the actual designator.

So if there is a type conversion on the input then the constraints for the port must come from the type mark not the signal being converted, but `unsigned` is unconstrained. There's an IEEE GitLab ticket about this: https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/294

(I don't know of any other way to achieve this.)

nickg, 2024-01-17:
Found it eventually. It comes from this paragraph in 8.1 of the 2008 LRM:

> Certain forms of name (indexed and selected names, slice names, and attribute names) include a prefix that is a name or a function call. If the prefix of a name is a function call, then the name denotes an element, a slice, or an attribute, either of the result of the function call, or (if the result is an access value) of the object designated by the result.

So the prefix of a sliced name can only be a function call or another name, and a type conversion is neither.

commit 4a7500e5 (2024-01-17): Check prefix of indexed or slice name is a name or function call

Fixes #835
closed 2024-01-17 by nickg