nickg/nvc #836: How to compile vendor libraries with different standard

m-kru, 2024-01-16:
By default vendor libraries are analyzed with the 2008 standard. I have some description where I require VHDL 2019 and I also utilize UVVM in test benches. However, during the analysis I get following error:
```
/home/user/.nvc/lib/uvvm_util/UVVM_UTIL.UVVM_UTIL_CONTEXT: design unit depends on 2008 version of STD.STANDARD but conflicting
          2019 version has been loaded
``` 
I have tried installing with the following command `nvc --std=2019 --install uvvm`, but it looks like the `--std` arguments is ignored by the `--install` command.

m-kru, 2024-01-17:
It looks like nvc simply calls scripts from the `contrib` directory. The script for UVVM has fixed std to 2008. Probably `nvc` should pass the command line argument for `--std` to the shell script.

nickg, 2024-01-17:
> The script for UVVM has fixed std to 2008. Probably `nvc` should pass the command line argument for `--std` to the shell script.

I've done this plus changed the UVVM script to build for both 2008 and 2019 by default.

closed 2024-01-17 by nickg

m-kru, 2024-01-17:
As it is possible to install vendor libraries directly executing the shell script I started wondering what is the point of the `--install` command? Doesn't it add unnecessary complexity to the nvc code?

nickg, 2024-01-17:
The fact they are implemented as shell scripts is an implementation detail. They will probably be rewritten in TCL at some point.

commit 4693b69a (2024-01-17): Allow the --std option to control which standard the install scripts use

Fixes #836