nickg/nvc #837: Analyzing vhdl file results in 'Fatal: Write to object in frozen arena'

augustofg, 2024-01-16:
Executing:
```bash
$ nvc --work=work --std=2008 -a dot_prod_pkg.vhd
$ nvc --work=work --std=2008 -a dot_prod.vhd
```
Results in nvc crashing with the following backtrace (files provided in the attachment):
```
** Fatal: Write to object in frozen arena WORK.DOT_PROD [address=0x7f471fdf0860]
[0x558816288891] 
[0x5588161d132d] 
[0x558816242274] 
[0x5588161d1dd8] 
[0x7f472476f70f] (/usr/lib/libc.so.6) 
[0x558816215ce2] 
[0x55881620e215] 
[0x55881620e509] 
[0x55881620e944] 
[0x5588162101df] 
[0x5588161ea92e] 
[0x5588161f5ea5] 
[0x5588162029e9] 
[0x558816203bd7] 
[0x55881624113a] 
[0x5588161cb7b8] 
[0x5588161c567b] 
[0x7f4724758ccf] (/usr/lib/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x7f4724758d89] (/usr/lib/libc.so.6) ../csu/libc-start.c:360 __libc_start_main@@GLIBC_2.34
[0x5588161c6be4] 
```
NVC version:
```
nvc 1.12-devel (1.11.0.r47.g3bf2ff2f) (Using LLVM 16.0.6)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```
[nvc-bug.tar.gz](https://github.com/nickg/nvc/files/13954681/nvc-bug.tar.gz)

augustofg, 2024-01-18:
Backtrace with debug information included:
```
nvc --work=work --std=2008 -a  ../../../modules/fofb_processing/dot_prod_pkg.vhd
nvc --work=work --std=2008 -a  ../../../modules/fofb_processing/dot_prod.vhd
** Fatal: Write to object in frozen arena WORK.DOT_PROD [address=0x7f53761f0860]
[0x55f4d35e87c6] ../src/diag.c:1016 diag_femit
[0x55f4d35e88ab] ../src/diag.c:1040 diag_emit
[0x55f4d355188e] ../src/util.c:585 fatal_trace
[0x55f4d35aeb38] ../src/object.c:358 check_frozen_object_fault
[0x55f4d3553d66] ../src/util.c:866 signal_handler
[0x7f537ab4a70f] (/usr/lib/libc.so.6) 
[0x55f4d3582944] ../src/tree.c:722 tree_set_flag
[0x55f4d359166d] ../src/sem.c:3135 sem_check_fcall.lto_priv.0
[0x55f4d359c34c] ../src/sem.c:6658 sem_check
[0x55f4d35886dd] ../src/sem.c:194 sem_check_range
[0x55f4d3588940] ../src/sem.c:237 sem_check_discrete_range.lto_priv.0
[0x55f4d3589033] ../src/sem.c:377 sem_check_constraint
[0x55f4d358910c] ../src/sem.c:409 sem_check_subtype_helper
[0x55f4d358920e] ../src/sem.c:445 sem_check_subtype.lto_priv.0
[0x55f4d358a0a1] ../src/sem.c:954 sem_check_signal_decl.lto_priv.0
[0x55f4d359c2ec] ../src/sem.c:6645 sem_check
[0x55f4d356b9d6] ../src/parse.c:7031 p_signal_declaration
[0x55f4d3571da6] ../src/parse.c:9112 p_block_declarative_item.lto_priv.0
[0x55f4d35746bd] ../src/parse.c:12456 p_architecture_declarative_part
[0x55f4d3579e91] ../src/parse.c:12519 p_architecture_body
[0x55f4d357a725] ../src/parse.c:12725 p_secondary_unit
[0x55f4d357a899] ../src/parse.c:12751 p_library_unit
[0x55f4d357aaf1] ../src/parse.c:12814 p_design_unit
[0x55f4d357ab73] ../src/parse.c:12865 parse
[0x55f4d35aceda] ../src/common.c:2332 analyse_file
[0x55f4d35573cc] ../src/nvc.c:239 analyse
[0x55f4d3557098] ../src/nvc.c:1907 process_command
[0x55f4d3559100] ../src/nvc.c:2047 main
```

nickg, 2024-01-18:
Should be fixed now, thanks for reporting.

commit 8f450ece (2024-01-18): Avoid crash with 'SUBTYPE attribute. Fixes #837
commit 0101f91f (2024-01-28): Avoid crash with 'SUBTYPE attribute. Fixes #837
commit bd818305 (2024-03-29): Refactor calculation of aggregate bounds. Fixes #837
closed 2024-01-18 by nickg