nickg/nvc #839: What does missing vcode unit during elaboration mean

m-kru, 2024-01-17:
I try to port some medium complexity (4 libraries, 11 files in total) test bench to use nvc. All the description is valid VHDL description and the test bench should pass (it passes with other simulators). However, during the elaboration nvc fails with following message:

```
** Fatal: missing vcode unit GENERAL_CORES.WISHBONE_PKG.T_WISHBONE_MASTER_IN_ARRAY$resolved
```
What does this mean? The `T_WISHBONE_MASTER_IN_ARRAY` is present in the `WISHBONE_PKG` and the package is present in the `GENERAL_CORES` library. The order of file analysis is also correct.

nickg, 2024-01-17:
Is the code public, or are you able to provide a minimal reproducer?

m-kru, 2024-01-17:
The code is public, but there are some extra dependencies in the build flow. I will try to prepare simple reproducer tomorrow.

m-kru, 2024-01-18:
Reproducer: https://github.com/m-kru/nvc-reproducer

commit e7d7acd6 (2024-01-18): Fix error getting resolved value of array-of-record signal

Fixes #839
commit 36e22885 (2024-01-28): Fix error getting resolved value of array-of-record signal

Fixes #839
closed 2024-01-18 by nickg