nickg/nvc #840: Caught signal 11 (SEGV_MAPERR)

m-kru, 2024-01-19:
```
   Process (init)
	 /home/mkru/workspace/FBDL/go-vfbdb/tests/co-simulations/common/vhdl-wb3/general-cores/fusesoc/../modules/wishbone/wb_crossbar/xwb_crossbar.vhd:128
** Warning: cannot represent multidimensional arrays in FST format
     > /home/mkru/workspace/FBDL/go-vfbdb/tests/co-simulations/common/vhdl-wb3/general-cores/fusesoc/../modules/wishbone/wb_crossbar/xwb_crossbar.vhd:139
     |
 139 |   signal matrix_old : matrix := (others => (others => '0'));
     |          ^^^^^^^^^^

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x5609230bd536] ***

[0x56092315ad5d] ../src/util.c:870 signal_handler
[0x7f2d7df4008f] (/usr/lib/x86_64-linux-gnu/libc-2.31.so)
[0x5609230bd536] ../src/tree.c:273 fst_type_for
[0x5609230bd536] ../src/rt/wave.c:250 fst_type_for
[0x5609230c117b] ../src/rt/wave.c:538 fst_process_signal
[0x5609230c3ac8] ../src/rt/wave.c:816 fst_walk_design.lto_priv.0
[0x5609230c3b0c] ../src/rt/wave.c:825 fst_walk_design.lto_priv.0
[0x5609230c3b0c] ../src/rt/wave.c:825 fst_walk_design.lto_priv.0
[0x560923162abc] ../src/cov/cov-exclude.c:850 run_cmd
[0x560923162abc] ../src/rt/wave.c:1911 process_command
[0x560923073548] ../src/nvc.c:2047 main

Please report this bug at https://github.com/nickg/nvc/issues
```

nickg, 2024-01-19:
It's working now with the latest master branch but fails to find an input file:

```
** Warning: cannot represent multidimensional arrays in FST format
     > vhdl/general_cores/xwb_crossbar.vhd:139
     |
 139 |   signal matrix_old : matrix := (others => (others => '0'));
     |          ^^^^^^^^^^
** Fatal: 0ms+0: failed to open /tmp/python-to-vhdl-wb3: No such file or directory
    > vhdl/work/cosim.vhd:96
    |
 96 |       file_open(rd_pipe, read_fifo_path, read_mode);
    |       ^
   Procedure COSIM_INTERFACE [STRING, STRING, STD_LOGIC, T_WISHBONE_IF, T_WISHBONE_BFM_CONFIG]
         vhdl/work/cosim.vhd:96
   Process :tb_cosim:_p9
         vhdl/work/tb.vhd:47
```

closed 2024-01-19 by nickg

m-kru, 2024-01-20:
If fails to find a file because the software part is missing in the reproducer. I have tested the whole co-simulation testbench. It now works as expected.