nickg/nvc #843: ** Fatal: dst argument to map signal is not a signal

smaslovski, 2024-01-20:
Hi,

I've found that NVC crashes at elaboration stage with the vhdl code given futher below. Here is the complete crash report:
```
$ nvc --std=08 -a bug1.vhd   
$ nvc --std=08 -e tb

Name       WORK.TB.FFT_INST.STAGE_N.EVEN_FFT
Kind       instance
Context    WORK.TB.FFT_INST.STAGE_N
Blocks     10
Registers  128
Types      36
  WORK.FFT_PKG.DATA_TYPE                {[32] : 0..8, [32] : 0..8}
  WORK.FFT_PKG.DATA_TYPE$               {$<0..8>, $<0..8>}
Variables  6
  POW                                   // -2^31..2^31-1 => 0..2^31-1, constant
  DIM                                   // -2^31..2^31-1 => 1..2^31-1, constant
  W                                     // [*] : WORK.FFT_PKG.DATA_TYPE{} => WORK.FFT_PKG.DATA_TYPE{}, constant
  Y                                     // [*] : WORK.FFT_PKG.DATA_TYPE${}, signal
  X                                     // [*] : WORK.FFT_PKG.DATA_TYPE${}, signal
  i9                                    // #, temp
Begin
   0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
      r1 := package init WORK.FFT_PKG   // P<WORK.FFT_PKG>
      r2 := package init IEEE.FLOAT_PKG // P<IEEE.FLOAT_PKG>
      r3 := const 1                     // -2^31..2^31-1 => 1
      r4 := const 0                     // -2^31..2^31-1 => 0
      r5 := const 2147483647            // -2^31..2^31-1 => 2^31-1
      r6 := const 0                     // 0..1 => 0
      POW := store r3
      r7 := const 2                     // -2^31..2^31-1 => 2
      DIM := store r7
      r8 := var upref 2, W              // @<[*] : WORK.FFT_PKG.DATA_TYPE{}> => WORK.FFT_PKG.DATA_TYPE{}
      r9 := load indirect r8            // [*] : WORK.FFT_PKG.DATA_TYPE{} => WORK.FFT_PKG.DATA_TYPE{}
      r10 := fcall WORK.FFT_PKG.EVEN_ELEM(24WORK.FFT_PKG.DATA_VECTOR)24WORK.FFT_PKG.DATA_VECTOR r1, r9 // [*] : WORK.FFT_PKG.DATA_TYPE{} => WORK.FFT_PKG.DATA_TYPE{}
      W := store r10
      r11 := var upref 1, EVEN          // @<[*] : WORK.FFT_PKG.DATA_TYPE${}> => WORK.FFT_PKG.DATA_TYPE{}
      r12 := load indirect r11          // [*] : WORK.FFT_PKG.DATA_TYPE${} => WORK.FFT_PKG.DATA_TYPE{}
      r13 := load DIM                   // -2^31..2^31-1 => 1..2^31-1
      r14 := debug locus WORK.TB.elab-763 // D<>
      r15 := sub r13 - r3               // -2^31..2^31-1 => 0..2147483646
      r16 := null                       // @<WORK.FFT_PKG.DATA_TYPE{}>
      r17 := wrap r16 [r4 r15 r6]       // [*] : WORK.FFT_PKG.DATA_TYPE{}
      r18 := debug locus WORK.TB.elab-1393 // D<>
      r19 := range length left r4 right r15 dir r6 // #
      r20 := uarray len r12 dim 0       // #
      length check left r19 == right r20 locus r18
      r21 := unwrap r12                 // @<WORK.FFT_PKG.DATA_TYPE${}> => WORK.FFT_PKG.DATA_TYPE${}
      r22 := wrap r21 [r4 r15 r6]       // [*] : WORK.FFT_PKG.DATA_TYPE${}
      Y := store r22
      r23 := debug locus WORK.TB.elab-721 // D<>
      r24 := index X                    // @<[*] : WORK.FFT_PKG.DATA_TYPE${}> => [*] : WORK.FFT_PKG.DATA_TYPE${}
      r25 := alloc r19                  // @<WORK.FFT_PKG.DATA_TYPE${}> => WORK.FFT_PKG.DATA_TYPE{}
      r26 := wrap r25 [r4 r15 r6]       // [*] : WORK.FFT_PKG.DATA_TYPE${}
      r24 := store indirect r26
      r27 := const 0                    // # => 0
      i9 := store r27
      r28 := debug locus WORK.TB.elab-759 // D<>
      push scope locus r28
      jump 1
   1: r29 := load i9                    // #
      r30 := cmp r29 == r19
      cond r30 then 3 else 2
   2: r31 := array ref r25 offset r29   // @<WORK.FFT_PKG.DATA_TYPE${}> => WORK.FFT_PKG.DATA_TYPE{}
      r32 := array ref r16 offset r29   // @<WORK.FFT_PKG.DATA_TYPE{}>
      r33 := debug locus WORK.TB.elab-759 // D<>
      push scope locus r33
      r34 := record ref r32 field 0     // @<0..8> => 0..8
      r35 := record ref r31 field 0     // @<$<0..8>> => $<0..8>
[... 269 lines cut ...]
begin
  fft_inst :
    entity work.fft(recursive)
      generic map (POW => POW)
      port map (x => x, y => y);
  test_proc :
    process is
      alias s is to_string [integer return string];
      alias s is to_string [real return string];
    begin
      for i in x'range loop
        x(i) <= data_type'(to_float(stimul(i).re), to_float(stimul(i).im));
      end loop;
      wait for 1 ps;
      for i in y'range loop
        assert abs(to_real(y(i).re) - result(i).re) < abstol and abs(to_real(y(i).im) - result(i).im) < abstol
          report "Inconsistent result detected: y(" & s(i) & ") = (" & s(to_real(y(i).re)) & "," & s(to_real(y(i).im)) & ")"  severity failure;
      end loop;
      wait;
    end process;
end architecture test;
```

BR,
Stanislav

smaslovski, 2024-01-20:
To add more to this report, I can tell that the crash does not appear if the code of fft module is chaged as follows:
```
architecture recursive of fft is
begin
  stage_n :
  if POW > 0 generate
    subtype half_array is data_vector (0 to DIM/2-1);
    signal x_e, x_o, even, odd : half_array;
    alias w_b : half_array is w(0 to DIM/2-1);
    alias w_t : half_array is w(DIM/2 to DIM-1);
  begin
    x_e <= even_elem(x);
    even_fft :
      entity fft
        generic map (
          POW => POW-1, w => even_elem(w))
        port map (
          x => x_e, y => even);
    x_o <= odd_elem(x);
    odd_fft :
      entity fft
        generic map (
          POW => POW-1, w => even_elem(w))
        port map (
          x => x_o, y => odd);
    butterfly:
    process (all) is
    begin
      for i in half_array'range loop
        y(i)       <= even(i) + odd(i) * w_b(i);
        y(i+DIM/2) <= even(i) + odd(i) * w_t(i);
      end loop;
    end process;
  end generate stage_n;
  stage_0 :
  if POW = 0 generate
    y <= x;
  end generate stage_0;
end architecture recursive;
```

In principle, it should be possible to achieve the same result just by adding "inertial" keyword before the function calls in the port map association lists. However, it seems that NVC does not know about such syntax, although it is a part of VHDL 2008 stantard.

BR,
Stanislav

nickg, 2024-01-26:
I've fixed the crash above and also added support for `inertial` in port maps.

closed 2024-01-26 by nickg