nickg/nvc #844: Simulation of large design ends with: Fatal: missing variable for generic X in Y

martinspinler, 2024-01-29:
Hello! Any quick hints to overcome this issue?
Unfortunately the code is not public.
```
Name       WORK.XXXXXXXXX.RX_HDR_ADDR_actual
Kind       process
Context    WORK.XXXXXXXXX
Blocks     2
Registers  27
Types      19
Variables  0
Begin
   0: r0 := var upref 1, RX_HDR_ADDR_actual // @<$<0..8>> => 0..8
      r1 := load indirect r0            // $<0..8> => 0..8
      r2 := const 2                     // # => 2
      r3 := const 1                     // # => 1
      r4 := const 15                    // # => 15
      r5 := const 30                    // # => 30
      drive signal r1 count r5
      r6 := var upref 2, RX_TRANS_OBUF_ADDR_REG // @<[*] : $<0..8>> => 0..8
      r7 := load indirect r6            // [*] : $<0..8> => 0..8
      r8 := uarray len r7 dim 0         // #
      r9 := unwrap r7                   // $<0..8>
      sched event on r9 count r8
      return 
   1: r10 := const 0                    // -2^63..2^63-1 => 0
      r11 := var upref 2, RX_TRANS_OBUF_ADDR_REG // @<[*] : $<0..8>> => 0..8
      r12 := load indirect r11          // [*] : $<0..8> => 0..8
      r13 := unwrap r12                 // $<0..8>
      r14 := resolved r13               // @<0..8> => 0..8
      r15 := uarray left r12 dim 0      // #
      r16 := uarray right r12 dim 0     // #
      r17 := uarray dir r12 dim 0       // 0..1
      r18 := wrap r14 [r15 r16 r17]     // [*] : 0..8
      r19 := var upref 1, MFB_REGIONS   // @<-2^31..2^31-1> => 0..2^31-1
      r20 := load indirect r19          // -2^31..2^31-1 => 0..2^31-1
      r21 := const 1                    // -2^31..2^31-1 => 1
      r22 := debug locus WORK.MFB_CROSSBARX_OUTPUT_BUFFER+808 // D<>
      r23 := sub r20 - r21              // -2^31..2^31-1 => -1..2147483646
      r24 := const 0                    // -2^31..2^31-1 => 0
      r25 := const 1                    // 0..1 => 1
      r26 := link package WORK.MATH_PACK // P<WORK.MATH_PACK>

** Fatal: missing variable for generic BUF_BYTES in WORK.MFB_CROSSBARX_OUTPUT_BUFFER
```

nickg, 2024-01-29:
Is is possible to share some details about the generic `buf_bytes` (e.g. its type)? `rx_hdr_addr` is presumably a port that's assigned some expression that uses `buf_bytes`, could you share e.g. the port map association?

martinspinler, 2024-02-16:
Thanks for the instant reply and sorry for my delay.
I have reproduced a very similar problem in open design, maybe we can find more info in this way.

If you don't mind, please clone the repository on github like below:
You don't need to have cocotb installed (I believe that you do anyway :) ), but you need to have make & tcl, also the Vivado & XPM libraries compiled, and also the dtc (Device Tree compiler, maybe the system package with name dtc/libfdt/libfdt-devel).

```
git clone https://github.com/CESNET/ndk-app-minimal.git
cd ndk-app-minimal
git submodule update --init ndk/ofm
git submodule update --init ndk/core
git submodule update --init ndk/cards

cd tests/cocotb;
make TARGET=nvc-sim -C ../../build/fb2cghh DMA_TYPE=4 PCIE_CONF=1xGen3x8LL TOP_LEVEL_ENT=fpga;
```

With nvc-git r1.11.0.r91.g3d4307e-1 on recent Arch Linux this brings me:
** Fatal: missing variable for generic AVST_DOWN_SEG in WORK.PCIE_ADAPTER

nickg, 2024-02-17:
Thanks for sharing this. It now gets as far as:

```
     0.00ns INFO     cocotb                             Seeding Python random module with 1708192803
     0.00ns ERROR    cocotb                             cocotb testbench initialization failed. Exiting.
                                                        Traceback (most recent call last):
                                                          File "/home/nick/src/cocotb/venv/lib/python3.11/site-packages/cocotb/__init__.py", line 207, in _initialise_testbench
                                                            _initialise_testbench_(argv_)
                                                          File "/home/nick/src/cocotb/venv/lib/python3.11/site-packages/cocotb/__init__.py", line 304, in _initialise_testbench_
                                                            regression_manager.discover_tests(modules=modules, filters=filters)
                                                          File "/home/nick/src/cocotb/venv/lib/python3.11/site-packages/cocotb/regression.py", line 192, in discover_tests
                                                            __import__(module_name)
                                                          File "/home/nick/src/cocotb/venv/lib/python3.11/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/nick/src/ndk-app-minimal/tests/cocotb/cocotb_test.py", line 6, in <module>
                                                            from ndk_core import NFBDevice
                                                        ModuleNotFoundError: No module named 'ndk_core'
```

I guess that's because there's some Python files missing from the repository. Could you try again with the latest master branch?

martinspinler, 2024-02-23:
That's perfect! Thank You, now it works.

closed 2024-02-23 by martinspinler