nickg/nvc #851: Crash when outputting fst

nathanaelhuffman, 2024-02-26:
I'm not sure what's the most useful, here to give you for additional debug, but happy to help get you what you need.

This is using vunit message passing (and vunit runner etc) and external names in the testbench, I can try to come up with a minimal reproducer if this isn't enough to go on.

If I don't generate waveforms, nvc runs to completion w/o crashing, but when I try to generate waveforms I get this crash:

```
*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x6213d476e2f4] ***

[0x6213d4704fed]
[0x70de8024251f] (/usr/lib/x86_64-linux-gnu/libc.so.6)
[0x6213d476e2f4]
[0x6213d47cf16d]
[0x6213d47d5ec8]
[0x6213d47d6eb2]
[0x6213d47d6dcc]
[0x6213d47d6dcc]
[0x6213d470f47c]
[0x6213d470b6a1]
[0x6213d470e134]
[0x6213d470b30f]
[0x6213d4701ec0]
[0x70de80229d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/nptl/libc_start_call_main.h:58 __libc_start_call_main
[0x70de80229e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../csu/libc-start.c:392 __libc_start_main@@GLIBC_2.34
[0x6213d47032a4]
```

nathanaelhuffman, 2024-02-26:
I generally consider any crash worth reporting, but it's possible I'm also doing something dumb in the testbench that's helping exhibit this misbehavior (notably my testcase isn't passing which is why i was going to look at waves).

If I run nvc with the --format=vcd option I get similar crashing behavior so it's maybe not fst-specific.

nickg, 2024-02-26:
Is it possible to get a stack trace from gdb? Running vunit with `run.py -v` should give you the full command line and then `gdb -ex run --args nvc ...` followed by `bt` when it crashes. Although it looks like the build you are using might not have debug symbols.

Can you post the first line of `nvc --version`?

nathanaelhuffman, 2024-02-26:
```
/usr/bin/nvc --version
nvc 1.11.3 (f90ac82) (Using LLVM 14.0.0)
Copyright (C) 2011-2023  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```
and yes there are no debug symbols in this build:

here's the backtrace:
```
(gdb) bt
#0  0x00005555555e52f4 in ?? ()
#1  0x000055555564616e in ?? ()
#2  0x000055555564cec9 in ?? ()
#3  0x000055555564deb3 in ?? ()
#4  0x000055555564ddcd in ?? ()
#5  0x000055555564ddcd in ?? ()
#6  0x000055555558647d in ?? ()
#7  0x00005555555826a2 in ?? ()
#8  0x0000555555585135 in ?? ()
#9  0x0000555555582310 in ?? ()
#10 0x0000555555578ec1 in ?? ()
#11 0x00007ffff0e29d90 in __libc_start_call_main (main=main@entry=0x555555577e60, argc=argc@entry=17,
    argv=argv@entry=0x7fffffffd098) at ../sysdeps/nptl/libc_start_call_main.h:58
#12 0x00007ffff0e29e40 in __libc_start_main_impl (main=0x555555577e60, argc=17, argv=0x7fffffffd098, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd088) at ../csu/libc-start.c:392
#13 0x000055555557a2a5 in ?? ()
```

I'll see if I can build locally with debug

nathanaelhuffman, 2024-02-26:
Ok I build HEAD here, and here's the backtrace with symbols, let me know if you need more info:
```
(gdb) bt
#0  folded_int (t=<optimized out>, l=0x7fffffffc5d0) at ../src/common.c:66
#1  0x000055555564b63e in fst_get_array_range (type=<optimized out>, cons=<optimized out>, s=<optimized out>,
    msb=0x7fffffffc6a0, lsb=0x7fffffffc698, length=0x7fffffffc6a8, dir=0x7fffffffc694) at ../src/rt/wave.c:452
#2  0x0000555555650fbc in fst_create_array_var (tb=0x55555666db90, cons=0x0, type=0x7fffe41500f0, s=0x7fffe7e00f80,
    d=0x7fffe4150220, wd=0x555555bda270) at ../src/rt/wave.c:483
#3  fst_process_signal (wd=0x555555bda270, scope=<optimized out>, d=0x7fffe4150220, cons=0x0, tb=0x55555666db90)
    at ../src/rt/wave.c:746
#4  0x0000555555651fd3 in fst_walk_design (wd=0x555555bda270, block=0x7fffe77f5080) at ../src/rt/wave.c:828
#5  0x0000555555651f1d in fst_walk_design (wd=0x555555bda270, block=0x7fffe77f4d00) at ../src/rt/wave.c:837
#6  0x0000555555651f1d in fst_walk_design (wd=0x555555bda270, block=0x7fffe77f0030) at ../src/rt/wave.c:837
#7  0x000055555557f55a in wave_dumper_restart (m=0x555556739e70, wd=0x555555bda270) at ../src/rt/wave.c:862
#8  run_cmd (argc=<optimized out>, argv=<optimized out>, state=<optimized out>) at ../src/nvc.c:839
#9  0x0000555555581462 in process_command (argc=4, argv=0x7fffffffd0b0, state=0x7fffffffcdf0) at ../src/nvc.c:1923
#10 0x0000555555583ee7 in elaborate (argc=4, argv=0x7fffffffd0b0, state=0x7fffffffcdf0) at ../src/nvc.c:521
#11 0x0000555555581190 in process_command (argc=9, argv=0x7fffffffd088, state=0x7fffffffcdf0) at ../src/nvc.c:1921
#12 0x0000555555577396 in main (argc=9, argv=0x7fffffffd088) at ../src/nvc.c:2059
```

nickg, 2024-02-27:
I managed to reproduce a similar crash with:

```vhdl
entity issue851 is
end entity;

architecture test of issue851 is
    signal s : bit_vector(1 to 3);
begin

    b: block is
        port ( p : in bit_vector );
        port map ( s );

        signal t : p'subtype;
    begin
    end block;

end architecture;
```

Could you test again with the latest master branch?

nathanaelhuffman, 2024-02-27:
master runs and generates waves successfully now, no crash, thanks very much!

closed 2024-02-27 by nickg