nickg/nvc #852: Array data not written to the FST file

m-kru, 2024-02-27:
I have some testbench with following declarative part:

```vhdl
architecture test of tb_writeb is

  signal clk : std_logic := '1';

  signal ck : checker_t := init;
  signal iface : interface_t := init;

  constant ADDR : unsigned(31 downto 0) := x"00000000";
  constant DATA : data_array_t := (
    x"01234567", x"89ABCDEF", x"DAEDBEEF", x"F0F0F0F0"
  );

  signal written_data : data_array_t(0 to 3);

  signal write_done : boolean := false;

begin
```

The `data_arary_t` is defined in a separate package as `type data_array_t is array (natural range <>) of std_logic_vector(31 downto 0);`. In the output .fst file I cannot find `ADDR`, `DATA`, and `written_data`. `ADDR` and `DATA` are constants so their absence is not a big deal. However, the `written_data` is a signal and it would be really nice to have a possibility to display it on a waveform.

nickg, 2024-02-27:
The `--dump-arrays` option should do that:

```
     --dump-arrays
             Include memories and nested arrays in the waveform data.  This is
             disabled by default as it can have significant performance, mem‐
             ory, and disk space overhead.
```

m-kru, 2024-02-27:
It indeed works. However, I observe some weird behavior.

```vhdl

  type data_array_t is array (natural range <>) of std_logic_vector(31 downto 0);

  type mock_completer_t is record
    -- Configuration elements
    prefix : string; -- Optional prefix used in report messages.
    -- Internal elements
    memory : data_array_t;
    -- Statistics elements
    read_count  : natural; -- Number of read transfers.
    write_count : natural; -- Number of write transfers.
  end record;

  function init (memory_size: natural; prefix: string := "apb: mock completer: ") return mock_completer_t is
    variable mc : mock_completer_t(prefix(0 to prefix'length-1), memory(0 to memory_size - 1));
  begin
    mc.prefix := prefix;
    return mc;
  end function;

  signal mc : mock_completer_t := init(memory_size => 8);
```

I would expect the `mc.memory` array to have 8 elements. However, the gtkwave shows 256 elements.

[image]

nickg, 2024-02-27:
> 
> I would expect the `mc.memory` array to have 8 elements. However, the gtkwave shows 256 elements.
> 

Yes this is a bug.

commit c258c7cf (2024-02-28): Incorrect record field length in FST dump. Fixes #852